armbian/build #1139: Support sqfs ROOTFS_TYPE

probonopd, 2018-10-23:
This is a feature request to support squashfs (`ROOTFS_TYPE=sqfs`) as an additional target format.

This would be extremely useful for embedded systems, since the squashfs filesystem is read-only and cannot inadvertently be altered or corrupted.

A read-write filesystem can be optionally overlaid to give the impression of a read-write filesystem, if so desired. However, it is always easy to go back to a "known good" factory-new state by discarding the read-write overlay.

Systems like OpenWrt and Volumio have been using this scheme for quite some time.

Example implementation:
https://github.com/volumio/Build/blob/master/scripts/initramfs/init_armbian

probonopd, 2018-10-25:
> I suggest looking into brtfs

Not an option for me because it does not provide the same one-readonly-file peace of mind and my personal experience with btrfs so far has been less than stellar in terms of robustness.

Related, OpenEmbeeded/Yocto does support squashfs as one of its `IMAGE_TYPES`.

ThomasKaiser, 2018-10-25:
> OpenEmbeeded/Yocto does support squashfs as one of its `IMAGE_TYPES`.

Great, so why not using this as embedded firmware then? 'Armbian' is obviously not an embedded firmware but just a build system allowing to run upstream Debian based distros in an optimized way on SBC. This includes the ability to upgrade packages in a normal way, and adding tons of complexity is obviously no good idea.

The Volumio approach is a great example since Volumio guys care only about a very limited subset of Armbian (they only support a few boards and only one specific kernel variant each which is not Armbian reality with tons of boards, up to 3 kernels per board and Igor encouraging users to switch between kernel versions on a daily basis with his `armbian-config` tool)

daiaji, 2018-11-05:
@probonopd Although I did not apply btrfs to production work.
But I use btrfs on my PC
And use btrfs to create RAID0 for two bad mechanical drives.
My VPS is also using btrfs
There is nothing catastrophic file system corruption due to insufficient robustness.
And in the opensuse forum, I have hardly heard of a problem caused by insufficient robustness of btrfs (even if the opensuse distribution uses btrfs as the system file system by default).
But it seems that the availability of RAID5-6 is still unclear (although this year seems to be rewritten by RAID5-6).
@ThomasKaiser In fact, because the read and write performance of btrfs is not good, I prefer to use F2FS on eMMC.
But the transparent compression in the F2FS development plan has not been progressing.

ThomasKaiser, 2018-11-05:
> poor read and write performance of BTRFS

What's the source for this?

daiaji, 2018-11-06:
@ThomasKaiser My benchmark result is that the btrfs file system usually uses lzo transparent compression. The write performance will be three-quarters of the ext4 file system.
Btrfs write performance without transparent compression is even worse.
Read performance is not so bad.

https://www.phoronix.com/scan.php?page=article&item=linux-415-fs
The phoronix benchmark will make people feel more disappointed with the performance of BTRFS.

probonopd, 2018-11-07:
My use case is "make the system resettable to a known-good default state".

igorpecovnik, 2018-11-07:
What about Ubuntu Bionic + overlayroot ([armbian-config]?(https://github.com/armbian/config/blob/master/debian-config-jobs#L1374-L1395))

probonopd, 2018-11-07:
Thanks for the pointer, yes, that's what I am currently using.

liftoff-sr, 2018-12-20:
@probonopd: this is probably not a lot of work.  You'd just have to redo a portion of the image creation logic.  Short of that DIY solution: I don't know what your volume is.  If it is high, then you can justify hiring somebody.  If it is low, then you can probably write a u-boot script to re-grab a golden "read-only" image off of a TFTP server when yours gets corrupted.  The copy on the TFPT server is "read-only" and safe from corruption.

lanefu, 2019-06-26:
closing old issue

closed 2019-06-26 by lanefu

probonopd, 2019-06-27:
Has this been implemented @lanefu? Not doing anything is usually not a good justification for closing feature requests...

lanefu, 2019-06-28:
@probonopd No developer or contribute has stepped up to take interest in the issue and consequently there is no plan to track or implement.   A motivated individual can still find this feature request by searching through the closed issues.  It's not deleted.    A pull request is welcome to be reviewed at anytime.