armbian/build #1140: Install and configure avahi-daemon by default

probonopd, 2018-10-23:
This is a feature request to make Armbian systems announce themselves in the network by default.

This would be _very_ useful since arguably many Armbian systems are headless and are connected to a network. To access them, the user then normally has to 
* Find out the IP address of the router, access the router as an administrator, and find out the IP address of the device from the list of assigned DHCP addresses (impossible if you have no administrative access to the router); or
* arp-scan the network for the MAC address of the device (impossible if you don't know the MAC address of the device and cumbersome if you do)

The solution would be to preinstall `avahi-daemon` and announce basic services by default:

```
apt install -y avahi-daemon
cd /etc/avahi/services/
wget -c https://raw.githubusercontent.com/lathiat/avahi/master/avahi-daemon/sftp-ssh.service
wget -c https://raw.githubusercontent.com/lathiat/avahi/master/avahi-daemon/ssh.service
cd -
service avahi-daemon restart
```

Now it can be accessed, e.g., from Linux desktops under Network as a "file share".

ThomasKaiser, 2018-10-24:
https://forum.armbian.com/topic/910-avahi/?do=findComment&comment=7031

Please provide for every distro variant Armbian currently supports the list of additional packages being installed and their total size.

ThomasKaiser, 2018-10-24:
> This would be very useful since arguably many Armbian systems are headless and are connected to a network. To access them, the user then normally has to

Not if the network administrator did his homework and set up an DHCP/DNS server combo capable of dynamic DNS updates. When I boot e.g. an NanoPi NEO4 the first time all I do after 15 seconds is

    ssh root@nanopineo4

and am in. Works without Zeroconf if DHCP/DNS server are set up appropriately (which should be default these days even with consumer routers).

But of course this only works once since for reasons unknown to me [this commit](https://github.com/armbian/build/commit/6fbb6b877797dcfbe7bb629ab0f3192ff848054c#diff-f8921cbfdf46404f614ffb2572b6ba91) destroyed everything and after the next reboot the board gets a new MAC address and the former DNS registration points to nowhere. 

Fortunately the commit message clearly describes why this destroyed functionality is needed (just kidding -- it simply sucks)

igorpecovnik, 2018-10-24:
> for reasons unknown to me this commit destroyed everything and after the next reboot the board gets a new MAC address and the former DNS registration points to nowhere.

Problem is random mac at mainline kernel (RK3399 in this case) and this workaround fix it. Now you should always get the same IP address from DHCP server. This shouldn't make thing worse. If it does, we need to find a better solution - somebody needs to dive into the kernel or u-boot driver code ...

ThomasKaiser, 2018-10-24:
> Problem is random mac at mainline kernel (RK3399 in this case) and this workaround fix it. Now you should always get the same IP address from DHCP server. This shouldn't make thing worse

Of course it did as explained **directly above**. The board boots with one specific MAC address, this MAC address will be used to get a DHCP lease (and updates the router's DNS table since the board submits its hostname as part of DHCP negotiation). In the meantime for whatever reasons the `set_fixed_mac` function calls `get_random_mac` (why not `use_actual_mac`?!) to assign **a totally different** random MAC address which will then be used subsequentially. This of course breaks the dynamic DNS update functionality in every network.

When testing for [A2 card performance](https://github.com/ThomasKaiser/Knowledge/blob/master/articles/A1_and_A2_rated_SD_cards.md) in an automated fashion a while ago I lost a lot of time getting the idea why boards that worked perfectly fine just some days ago with the same MAC address all the time and regardless how many times I flash an image (RK default kernel) will now have **random** MAC adresses after the first reboot.

If you want to force a MAC address why can't you choose the one that's already in use? This 'choose a random MAC address' is totally unneeded and breaks functionality like these DHCP/DNS interaction.

igorpecovnik, 2018-10-24:
> why not use_actual_mac

Yes, that would be better. Will check if that is possible. Anyway, RK3399 was fixed upstream and workaround will be removed in next commit along with some config cleanup ...

ThomasKaiser, 2018-10-24:
Why should another ugly hack like this

    CONNECTION="$(nmcli -f UUID,ACTIVE,DEVICE,TYPE connection show --active | tail -n1)"
    UUID=$(awk -F" " '/ethernet/ {print $1}' <<< "${CONNECTION}")
    DEVNAME=$(awk -F" " '/ethernet/ {print $3}' <<< "${CONNECTION}")
    MACADDR=$(/sbin/ip link | grep -A1 ${DEVNAME} | awk -F" " '/ether / {print $2}')

don't work? 

But why does stuff like this gets applied to whole board families in the first place and not just to specific affected family/kernel combinations (`dev`!) clearly marked as a workaround to be removed once the issue has been resolved? 

And of course something like this will break on boards with various NICs (EspressoBin) and of course the usual sanity checks are needed.

igorpecovnik, 2018-10-24:
It will break on Espressobin for different reason - Network manager is not in action. It's a workaround as the title of this section tells: https://github.com/armbian/build/blob/master/packages/bsp/common/usr/lib/armbian/armbian-firstrun#L86-L127

I will do some test to see if it needs further adjustments.

ThomasKaiser, 2018-10-24:
> It's a workaround as the title of this section tells

Why does it 'need' to be executed on systems where everything works flawlessly?! If it's a `dev` branch issue why not making this a requirement to fiddle around in NM config then?

igorpecovnik, 2019-05-31:
> Please provide for every distro variant Armbian currently supports the list of additional packages being installed and their total size.

Bionic:
```
Need to get 163 kB of archives.
After this operation, 751 kB of additional disk space will be used.
Get:1 http://ports.ubuntu.com bionic-security/main arm64 libavahi-common-data arm64 0.7-3.1ubuntu1.2 [22.1 kB]
Get:2 http://ports.ubuntu.com bionic-security/main arm64 libavahi-common3 arm64 0.7-3.1ubuntu1.2 [18.4 kB]
Get:3 http://ports.ubuntu.com bionic-security/main arm64 libavahi-core7 arm64 0.7-3.1ubuntu1.2 [66.1 kB]
Get:4 http://ports.ubuntu.com bionic-security/main arm64 avahi-daemon arm64 0.7-3.1ubuntu1.2 [56.1 kB]
```

Stretch:
```
Need to get 345 kB of archives.
After this operation, 1405 kB of additional disk space will be used.
Get:1 http://cdn-fastly.deb.debian.org/debian stretch/main arm64 libavahi-common-data arm64 0.6.32-2 [118 kB]
Get:2 http://cdn-fastly.deb.debian.org/debian stretch/main arm64 libavahi-common3 arm64 0.6.32-2 [48.4 kB]
Get:3 http://cdn-fastly.deb.debian.org/debian stretch/main arm64 libavahi-core7 arm64 0.6.32-2 [94.5 kB]
Get:4 http://cdn-fastly.deb.debian.org/debian stretch/main arm64 avahi-daemon arm64 0.6.32-2 [83.9 kB]

```

armhf is almost the same size. Shell we add this?

ThomasKaiser, 2019-05-31:
I see 6 distro variants [here](https://github.com/armbian/build/blob/24c81d1b05eb479d5693f5b3544d6a3363cd5a95/lib/general.sh#L360) so why are 4 of them of no relevance any more?

Asides that I'm a big fan of both reasonable service propagation and Zeroconf and would even suggest to add `avahi-autoipd` package to make things complete. But since adding packages has a price and I still don't think too much users will later be able to benefit from I don't know whether it's worth the disk size.

Majority of users is still on Windows and the first Windows version able to deal with Zeroconf is Win10 (there even a simple `ssh root@nanopineo4.local` in `cmd.exe` works). But for this to work they need to know what to search for and if 'Keeping the board's MAC address' would be a design principle in Armbian majority of users would be fine doing `ssh root@nanopineo4` instead not needing avahi at all. But since they don't know that this works it makes no difference.

Same on macOS. The OS allows to browse for available Zeroconf SSH announcements:

    mac-tk-2018:~ tk$ dns-sd -B _ssh._tcp .
    Browsing for _ssh._tcp
    DATE: ---Fri 31 May 2019---
    10:15:39.469  ...STARTING...
    Timestamp     A/R    Flags  if Domain               Service Type         Instance Name
    10:15:39.470  Add        3  12 local.               _ssh._tcp.           mac-tk-2018
    10:15:39.470  Add        3  17 local.               _ssh._tcp.           mac-tk-2018
    10:15:39.470  Add        3  18 local.               _ssh._tcp.           mac-tk-2018
    10:15:39.471  Add        3  12 local.               _ssh._tcp.           lime-a10 - SSH
    10:15:39.471  Add        2  12 local.               _ssh._tcp.           RockPro64 - SSH
    ^C

But users don't know and as such this functionality is still 'hidden'. So in the end some Linux users will be the only ones that benefit from an Armbian board showing up with its name to be accessed via SSH.

If Armbian would switch to btrfs with compressed rootfs in situations with kernel 4.4 or above I wouldn't think a second about adding more packages. But with the above limited benefit in mind I don't think another MB wasted for those packages is ok.

BTW: a totally different story would be setting the hostname to `armbian` at image creation time since now Armbian could promote to simply use `ssh root@armbian` since this works in 99.x% of all network installations flawlessly (due to DHCP with dynamic DNS updates now being standard). But this would require keeping the MAC address over the device's lifetime and also encouraging users to adjust the hostname if they deal with more than Armbian install. 

With Zeroconf/avahi it would technically work to keep the same hostname since the hostname would be appended with a number (`armbian.local`, `armbian-1.local`, `armbian-2.local` and so on) but of course it makes no sense at all.

igorpecovnik, 2019-09-23:
This feature was added to the desktop builds, where size is not that critical and to the "armbian-config easy install". Let's close this issue for now. We don't have consensus for placing this feature anywhere else.

closed 2019-09-23 by igorpecovnik