armbian/build #1141: USB-C Hub not detected on NanoPC-T4 (RK3399) Dev kernel

carlosedp, 2018-10-24:
I have a Sinstar USB-C Hub (https://www.amazon.com/USB-Hub-Sinstar-Aluminum-Ethernet/dp/B073VKGFBM) that I plugged into my NanoPC-T4 running Armbian Dev version (Kernel 4.19.0).

The hub ports were not detected and a USB drive plugged into it was not detected as well.

To test its support, I connected it into a Firefly RK3399 running Rockchip's 4.4.138 Kernel and the hub was perfectly detected with all it's ports and anything connected to it.

This might be some CONFIG missing for the Kernel or some patching from Rockchip. I checked that the param `Symbol: USB_XHCI_HCD [=y]` is enabled on the used kernel.

Here is the log for the Armbian Dev:
```
[ 1331.941391] fusb302 4-0022: CC connected in CC1 as DFP
[ 1331.941391] fusb302 4-0022: CC connected in CC1 as DFP
[ 1332.202205] fusb302 4-0022: PD connected as DFP, supporting 5V
[ 1332.202205] fusb302 4-0022: PD connected as DFP, supporting 5V
[ 1332.241177] fusb302 4-0022: DP config successful, pin_assignment 0x8
[ 1332.241177] fusb302 4-0022: DP config successful, pin_assignment 0x8
```

Here is the log for my Firefly RK3399 that detected it:
```
[867958.364379] fusb302 4-0022: CC connected in CC1 as DFP
[867958.386000] xhci-hcd xhci-hcd.7.auto: xHCI Host Controller
[867958.386593] xhci-hcd xhci-hcd.7.auto: new USB bus registered, assigned bus number 7
[867958.386814] xhci-hcd xhci-hcd.7.auto: hcc params 0x0220fe64 hci version 0x110 quirks 0x02030010
[867958.386867] xhci-hcd xhci-hcd.7.auto: irq 232, io mem 0xfe800000
[867958.391607] usb usb7: New USB device found, idVendor=1d6b, idProduct=0002
[867958.391632] usb usb7: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[867958.391648] usb usb7: Product: xHCI Host Controller
[867958.391662] usb usb7: Manufacturer: Linux 4.4.138 xhci-hcd
[867958.391675] usb usb7: SerialNumber: xhci-hcd.7.auto
[867958.393215] hub 7-0:1.0: USB hub found
[867958.393289] hub 7-0:1.0: 1 port detected
[867958.394118] xhci-hcd xhci-hcd.7.auto: xHCI Host Controller
[867958.394208] pwm-regulator vdd-log: Failed to get PWM: -517
[867958.396593] xhci-hcd xhci-hcd.7.auto: new USB bus registered, assigned bus number 8
[867958.396707] usb usb8: We don't know the algorithms for LPM for this host, disabling LPM.
[867958.396832] usb usb8: New USB device found, idVendor=1d6b, idProduct=0003
[867958.396843] usb usb8: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[867958.396852] usb usb8: Product: xHCI Host Controller
[867958.396860] usb usb8: Manufacturer: Linux 4.4.138 xhci-hcd
[867958.396868] usb usb8: SerialNumber: xhci-hcd.7.auto
[867958.398029] hub 8-0:1.0: USB hub found
[867958.398075] hub 8-0:1.0: 1 port detected
[867958.398593] pwm-regulator vdd-log: Failed to get PWM: -517
[867958.398605] rockchip-dwc3 usb@fe800000: USB HOST connected
[867958.652803] fusb302 4-0022: PD connected as DFP, supporting 5V
[867958.734952] fusb302 4-0022: DP config successful, pin_assignment 0x8
[867962.367998] usb 8-1: new SuperSpeed USB device number 2 using xhci-hcd
[867962.381741] usb 8-1: New USB device found, idVendor=2109, idProduct=0817
[867962.381756] usb 8-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[867962.381763] usb 8-1: Product: USB3.0 Hub             
[867962.381769] usb 8-1: Manufacturer: VIA Labs, Inc.         
[867962.397277] hub 8-1:1.0: USB hub found
[867962.397660] hub 8-1:1.0: 4 ports detected
[867962.399680] pwm-regulator vdd-log: Failed to get PWM: -517
[867962.482497] usb 7-1: new high-speed USB device number 2 using xhci-hcd
[867962.597985] usb 7-1: New USB device found, idVendor=2109, idProduct=2817
[867962.598000] usb 7-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[867962.598007] usb 7-1: Product: USB2.0 Hub             
[867962.598013] usb 7-1: Manufacturer: VIA Labs, Inc.         
[867962.621233] hub 7-1:1.0: USB hub found
[867962.621410] hub 7-1:1.0: 4 ports detected
[867962.779059] pwm-regulator vdd-log: Failed to get PWM: -517
[867962.943192] usb 8-1.2: new SuperSpeed USB device number 3 using xhci-hcd
[867962.968180] usb 8-1.2: New USB device found, idVendor=0bda, idProduct=0411
[867962.968196] usb 8-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[867962.968203] usb 8-1.2: Product: 4-Port USB 3.0 Hub
[867962.968209] usb 8-1.2: Manufacturer: Generic
[867962.976380] hub 8-1.2:1.0: USB hub found
[867962.977725] hub 8-1.2:1.0: 4 ports detected
[867962.982380] pwm-regulator vdd-log: Failed to get PWM: -517
[867962.983150] pwm-regulator vdd-log: Failed to get PWM: -517
[867963.047412] usb 8-1.4: new SuperSpeed USB device number 4 using xhci-hcd
[867963.065666] usb 8-1.4: New USB device found, idVendor=05e3, idProduct=0749
[867963.065692] usb 8-1.4: New USB device strings: Mfr=3, Product=4, SerialNumber=2
[867963.065707] usb 8-1.4: Product: USB3.0 Card Reader
[867963.065721] usb 8-1.4: Manufacturer: Generic
[867963.065734] usb 8-1.4: SerialNumber: 000000001536
[867963.067833] usb-storage 8-1.4:1.0: USB Mass Storage device detected
[867963.069119] scsi host3: usb-storage 8-1.4:1.0
[867963.070406] pwm-regulator vdd-log: Failed to get PWM: -517
[867963.071192] pwm-regulator vdd-log: Failed to get PWM: -517
[867963.248746] usb 8-1.2.1: new SuperSpeed USB device number 5 using xhci-hcd
[867963.261360] usb 8-1.2.1: New USB device found, idVendor=0bda, idProduct=8153
[867963.261376] usb 8-1.2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=6
[867963.261382] usb 8-1.2.1: Product: USB 10/100/1000 LAN
[867963.261388] usb 8-1.2.1: Manufacturer: Realtek
[867963.261394] usb 8-1.2.1: SerialNumber: 001000001
[867963.278639] pwm-regulator vdd-log: Failed to get PWM: -517
[867963.383498] usb 7-1.1: new high-speed USB device number 3 using xhci-hcd
[867963.540436] usb 7-1.1: New USB device found, idVendor=2109, idProduct=0102
[867963.540451] usb 7-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[867963.540458] usb 7-1.1: Product: USB 2.0 BILLBOARD             
[867963.540465] usb 7-1.1: Manufacturer: VIA Technologies Inc.         
[867963.540471] usb 7-1.1: SerialNumber: 0000000000000001
[867963.549615] pwm-regulator vdd-log: Failed to get PWM: -517
[867963.597012] usb 8-1.2.1: reset SuperSpeed USB device number 5 using xhci-hcd
[867963.613195] r8152 8-1.2.1:1.0 (unnamed net_device) (uninitialized): Unknown version 0x6010
[867963.613211] r8152 8-1.2.1:1.0 (unnamed net_device) (uninitialized): Unknown Device
[867963.731573] usb 7-1.2: new high-speed USB device number 4 using xhci-hcd
[867963.883525] usb 7-1.2: New USB device found, idVendor=0bda, idProduct=5411
[867963.883541] usb 7-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[867963.883548] usb 7-1.2: Product: 4-Port USB 2.0 Hub
[867963.883554] usb 7-1.2: Manufacturer: Generic
[867963.901704] hub 7-1.2:1.0: USB hub found
[867963.902607] hub 7-1.2:1.0: 4 ports detected
[867963.935792] pwm-regulator vdd-log: Failed to get PWM: -517
[867964.073798] scsi 3:0:0:0: Direct-Access     Generic  MassStorageClass 1536 PQ: 0 ANSI: 6
[867964.076239] scsi 3:0:0:1: Direct-Access     Generic  MassStorageClass 1536 PQ: 0 ANSI: 6
[867964.079390] pwm-regulator vdd-log: Failed to get PWM: -517
[867964.080999] pwm-regulator vdd-log: Failed to get PWM: -517
[867964.486360] sd 3:0:0:0: [sdc] Attached SCSI removable disk
[867964.490382] sd 3:0:0:1: [sdd] Attached SCSI removable disk
```

Connection of a USB drive:
```
[867979.577616] usb 7-1.2.2: new high-speed USB device number 5 using xhci-hcd
[867980.173251] usb 7-1.2.2: New USB device found, idVendor=054c, idProduct=0243
[867980.173331] usb 7-1.2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[867980.173346] usb 7-1.2.2: Product: Storage Media
[867980.173360] usb 7-1.2.2: Manufacturer: Sony
[867980.173373] usb 7-1.2.2: SerialNumber: 2A07082739728
[867980.174570] usb 7-1.2.2: ep 0x81 - rounding interval to 128 microframes, ep desc says 255 microframes
[867980.174620] usb 7-1.2.2: ep 0x2 - rounding interval to 128 microframes, ep desc says 255 microframes
[867980.175611] usb-storage 7-1.2.2:1.0: USB Mass Storage device detected
[867980.176980] scsi host4: usb-storage 7-1.2.2:1.0
[867980.178192] pwm-regulator vdd-log: Failed to get PWM: -517
[867980.178807] pwm-regulator vdd-log: Failed to get PWM: -517
[867981.183151] scsi 4:0:0:0: Direct-Access     Sony     Storage Media    0100 PQ: 0 ANSI: 0 CCS
[867981.186761] pwm-regulator vdd-log: Failed to get PWM: -517
[867981.188557] sd 4:0:0:0: [sde] 7864320 512-byte logical blocks: (4.03 GB/3.75 GiB)
[867981.189793] sd 4:0:0:0: [sde] Write Protect is off
[867981.189820] sd 4:0:0:0: [sde] Mode Sense: 43 00 00 00
[867981.190684] sd 4:0:0:0: [sde] No Caching mode page found
[867981.190707] sd 4:0:0:0: [sde] Assuming drive cache: write through
[867981.196531]  sde: sde1
[867981.200222] sd 4:0:0:0: [sde] Attached SCSI removable disk
```

igorpecovnik, 2018-10-25:
https://github.com/armbian/build/commit/a9f6207a3e50d132878bff6aae11cd5d5d1e158f#diff-db9469e2c69f3c276d7ccfc053ceea7d

This is build script issue tracker - here is the place for such discussion/notice: https://forum.armbian.com/forum/4-development DEV kernels are usually months away from any serious involvement.

closed 2018-10-25 by igorpecovnik

carlosedp, 2018-10-25:
Hi Igor, sorry about the noise. Gonna check the forums. Thanks!