armbian/build #1144: Does the nand_sata_install.sh needs to be cleaned?

malvcr, 2018-10-28:
_I am not sure if these notes belong to the build area.  If not, help me pointing where they must be.  Thank you._

As part of a bring up process, I was checking the nand_sata_install.sh script.  And there are some things that I think must be checked there.

- 1. Open Media Vault - specific instructions in the raw Armbian installation.  Check lines 19 and 290.  Could that be separated from the main, generic install script?  Although OMV it is a very close Armbian friend, it is not Armbian and doesn't match with all possible Armbian usage scenarios.
- 2. The script is trying to follow some logic for A10, A13 and A20 SOCs, but later change the methodology for the rk3328 ... that way of dealing with differences doesn't look coherent.  In fact, the script logic defines every SOC it is not an A10 or A13 to be an A20 one.  Even if that works, it is semantically wrong and could lead to future problems.
- 3. The running services (lines 124, 125) belong to a limited static set instead of trying to figure what are the real ones running in the corresponding OS partition, that could have different services running.

For the (i) could be possible to check some directory for "extra" add-on configurations instead of trying to include particular products with their behavior?

In the (ii), the A10, A13 and A20 way of reading parameters from a file looks better than wiring some information.  Also, could be possible to get the DEVICE_TYPE from a parameters file as the LINUXFAMILY, or to calculate it someway instead of wiring it inside a script that could be generic.

For (iii), as ARMBIAN has a specified method to declare services, seems better to query the system following that method instead of having a static written list that, probably, will not be complete and could lead to an incomplete copy procedure in case some files are used and/or mutating.

Thanks

igorpecovnik, 2018-10-29:
We already decide (some time ago) to rather start developing this script from scratch since it is messy and some parts could be abandoned. NAND support for example. I can't find where we talked about - we already exposed some other problems which should be fixed/done differently, and that should be brought up here. Idea is to leave this script as is, maintain it the best way possible, but start developing new one. That moment hasn't come yet. It should be easier than poking around this mess. (I am out for vacations so I won't be able to do much regarding this for another 2 weeks. Can only add some short inputs here and there.)

I would propose to move conversation, what and how shell be done, there - probably it was on forum, then opening a PR containing refactored/new/independent "armbian-install" script.

malvcr, 2018-10-29:
Let me swim in the forum, to see if I can find the conversation... or maybe to create a new one.

In general, if this is very messy and must be done from scratch, I think that the right focus it is not how to change it (historical background) but what must be done (actual needs).  Also, I was thinking that these scripts lack a unified information source that, in complete order, could grow from the uboot time to advanced requirements after the kernel is working.  That way won't be necessary to calculate things, even the same in different places, on each script. .... but this is a theme for the conversation :-)

Le me see how I can help, maybe to work this from an architectural point of view.

igorpecovnik, 2018-10-29:
> actual needs

Exactly.

Check this PR comments https://github.com/armbian/build/pull/1129 We also plan to move out u-boot flashing from current automatic to manual. To armbian-config or "nand-sata-install" script. Some parts of "update SPI flash" are already present and we have more boards where this is possible but is not taken care of.

malvcr, 2018-10-30:
An integral part must be how gather information.  Right now, it is an heterogeneous method with sources in different places, and even repeating the process in different scripts, making them difficult to maintain and, easier to break.  I think that a light and well behaved unified information source can simplify many tasks.  The /proc has some data, but it doesn't exist on uboot time, neither has the distribution specific things.   Let me start a conversation on that.

malvcr, 2018-10-30:
Done .... "Revisiting the installation process"  in the developer forum.  To see what happens :-)

igorpecovnik, 2018-11-03:
> has a censorship problem

Perhaps in a b/w world. If you think someone did abuse power, there are other ways to fix this. I am 100% on your side but since I haven't notice censorship in action I can't do anything about ... but if you say so I believe it happened.

lanefu, 2019-03-06:
closing old issue

closed 2019-03-06 by lanefu