armbian/build #1150: Add backport for the support for 802.1ad in the stmmac driver

ehoffman2, 2018-11-07:
Support for 802.1ad (S-VLAN + C-VLAN) in the stmmac driver has been added in mainline kernel 4.18.  I would suggest making a backport.

https://github.com/torvalds/linux/commit/ab188e8f4aad9845589ed050bde9514550a23ea5

Without this patch, the board is able to send 802.1ad packets, but on receive, there is a bug (or rather an incomplete support) that make the outer tag, the 0x88a8, being replaced with 0x8100.  So, basically the rest of the stack just see Q-in-Q packets.

To test the issue, I used an Orange PI PC2 (you can use any other boards based on ST-Micro Ethernet, i.e. using stmmac driver), and an Ubuntu 18.04 PC.  I create a C-VLAN 101 inside an S-VLAN 24.  On the board, the interface is eth0, and on my Ubuntu, my interface is eth1:

Configure 802.1ad on Orange PI PC2
----------------------------------
sudo ip link add link eth0 eth0.24 type vlan proto 802.1ad id 24
sudo ip link add link eth0.24 eth0.24.101 type vlan proto 802.1Q id 101
sudo ip link set eth0 up
sudo ip link set eth0.24 up
sudo ifconfig eth0.24.101 10.10.101.2/24

Configure 802.1ad on Linux PC
-----------------------------
sudo ip link add link eth1 eth1.24 type vlan proto 802.1ad id 24
sudo ip link add link eth1.24 eth1.24.101 type vlan proto 802.1Q id 101
sudo ip link set eth1 up
sudo ip link set eth1.24 up
sudo ifconfig eth1.24.101 10.10.101.1/24

Then, try to ping the other system, and you will see that the ARP requests on the board ingress will come out as Q-in-Q (using tcpdump in shell, or Wireshark in a GUI), i.e. both tags in the skbuf are seen as 0x8100.

With patch applied, the traffic flow through properly.

Regards,
Eric

lanefu, 2019-03-08:
@ehoffman2 

OPI PC2 build scripts have been on 4.19 for quite a while... Is this issue still relevant?

lanefu, 2019-03-20:
closing... no follow-up

closed 2019-03-20 by lanefu