armbian/build #1165: /etc/apt/apt.conf.d/02-armbian-compress-indexes incompatibility

ThomasKaiser, 2018-11-26:
Without doing an `rm /etc/apt/apt.conf.d/02-armbian-compress-indexes` there's an issue with installing the `apt-show-versions` package (webmin dependency). To me it looks like the bug is in apt-show-versions's `postinstall` script but it took me a little time to spot the difference when installing webmin the last days on a bunch of Bionic installs (those unaffected were using upstream x86 or ayufan installers).

igorpecovnik, 2018-11-26:
Se better remove this feature?

ThomasKaiser, 2018-11-26:
No idea. Filing an upstream bug report hoping for a fix would also be an option. I just did a google search, ended up [here](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=617856), read the date and immediately closed the tab.

Removing the file followed by `apt clean` and `apt update` 'fixed' it for me without wasting my time with ignorance and other unpleasant stuff.

Please note that I had no time to test on Debian too so no idea whether this only occurs with Ubuntu or not.

igorpecovnik, 2018-11-26:
Let be disabled by default. I also don't have any Bionic around so it will be enabled back there if it works.

closed 2018-11-26 by igorpecovnik