armbian/build #1170: Fresh build fails for OrangePiZero On Docker

nicopace, 2018-11-28:
Command run:
```
git clone --depth=1 https://github.com/armbian/build
cd build
./compile.sh docker BRANCH=next BOARD=orangepizero KERNEL_ONLY=no KERNEL_CONFIGURE=no BUILD_DESKTOP=no EXTERNAL=yes RELEASE=stretch
```

Expected result: get an image for orangepizero

Actual result: (last part of it, skipping kernel build that went ok)
```
dpkg-deb: building package 'linux-dtb-next-sunxi' in '../linux-dtb-next-sunxi_5.67_armhf.deb'.
dpkg-deb: building package 'linux-headers-next-sunxi' in '../linux-headers-next-sunxi_5.67_armhf.deb'.
dpkg-deb: building package 'linux-image-next-sunxi' in '../linux-image-next-sunxi_5.67_armhf.deb'.
dpkg-genchanges: warning: unknown substitution variable ${kernel:debarch}
dpkg-genchanges: warning: package linux-libc-dev-next-sunxi in control file but not in files list
dpkg-genchanges: info: binary-only upload (no source code included)
dpkg-deb: building package 'linux-source-4.14.84-next-sunxi' in '/root/armbian/.tmp/linux-source-next-sunxi_5.67_all.deb'.
[ o.k. ] Creating board support package [ orangepizero next ]
[ .... ] Fingerprinting 
[ o.k. ] Building package [ linux-stretch-root-next-orangepizero ]
[ o.k. ] Building desktop package [ armbian-stretch-desktop_5.67_all ]
[ o.k. ] Starting rootfs and image building process for [ orangepizero stretch ]
[ o.k. ] Creating new rootfs for [ stretch ]
[ o.k. ] Installing base system [ Stage 1/2 ]
I: Retrieving InRelease 
I: Retrieving Release 
I: Retrieving Release.gpg 
I: Checking Release signature
I: Valid Release signature (key id 067E3C456BAE240ACEE88F6FEF0F382A1A7B6500)
I: Retrieving Packages 
I: Validating Packages 
I: Resolving dependencies of required packages...
I: Resolving dependencies of base packages...
I: Found additional required dependencies: libaudit1 libaudit-common libbz2-1.0 libcap-ng0 libdb5.3 libdebconfclient0 libgcrypt20 libgpg-error0 liblz4-1 libncursesw5 libsemanage1 libsemanage-common libsystemd0 libudev1 libustr-1.0-1 
I: Found additional base dependencies: dmsetup gnupg-agent libapparmor1 libassuan0 libbsd0 libcap2 libc-l10n libcryptsetup4 libdevmapper1.02.1 libdns-export162 libelf1 libfastjson4 libffi6 libgmp10 libgnutls30 libhogweed4 libidn11 libidn2-0 libip4tc0 libip6tc0 libiptc0 libisc-export160 libksba8 liblocale-gettext-perl liblognorm5 libmnl0 libncurses5 libnetfilter-conntrack3 libnettle6 libnfnetlink0 libnpth0 libp11-kit0 libpsl5 libseccomp2 libsqlite3-0 libtasn1-6 libtext-charwidth-perl libtext-iconv-perl libtext-wrapi18n-perl libunistring0 libxtables12 pinentry-curses xxd 
I: Checking component main on http://httpredir.debian.org/debian...
I: Retrieving libacl1 2.2.52-3+b1
I: Validating libacl1 2.2.52-3+b1
I: Retrieving adduser 3.115
I: Validating adduser 3.115
I: Retrieving libapparmor1 2.11.0-3+deb9u2
I: Validating libapparmor1 2.11.0-3+deb9u2
I: Retrieving apt 1.4.8
I: Validating apt 1.4.8
I: Retrieving apt-utils 1.4.8
I: Validating apt-utils 1.4.8
I: Retrieving libapt-inst2.0 1.4.8
I: Validating libapt-inst2.0 1.4.8
I: Retrieving libapt-pkg5.0 1.4.8
I: Validating libapt-pkg5.0 1.4.8
I: Retrieving libattr1 1:2.4.47-2+b2
I: Validating libattr1 1:2.4.47-2+b2
I: Retrieving libaudit-common 1:2.6.7-2
I: Validating libaudit-common 1:2.6.7-2
I: Retrieving libaudit1 1:2.6.7-2
I: Validating libaudit1 1:2.6.7-2
I: Retrieving base-files 9.9+deb9u6
I: Validating base-files 9.9+deb9u6
I: Retrieving base-passwd 3.5.43
I: Validating base-passwd 3.5.43
I: Retrieving bash 4.4-5
I: Validating bash 4.4-5
I: Retrieving libdns-export162 1:9.10.3.dfsg.P4-12.3+deb9u4
I: Validating libdns-export162 1:9.10.3.dfsg.P4-12.3+deb9u4
I: Retrieving libisc-export160 1:9.10.3.dfsg.P4-12.3+deb9u4
I: Validating libisc-export160 1:9.10.3.dfsg.P4-12.3+deb9u4
I: Retrieving bsdmainutils 9.0.12+nmu1
I: Validating bsdmainutils 9.0.12+nmu1
I: Retrieving libbz2-1.0 1.0.6-8.1
[... 693 lines cut ...]
Get:12 http://cdn-fastly.deb.debian.org/debian stretch-backports/non-free armhf Packages [7117 B]
Get:13 http://cdn-fastly.deb.debian.org/debian stretch-backports/contrib armhf Packages [6597 B]
Get:14 http://cdn-fastly.deb.debian.org/debian stretch-backports/main armhf Packages [627 kB]
Get:15 http://cdn-fastly.deb.debian.org/debian stretch/contrib armhf Packages [49.6 kB]
Get:16 http://cdn-fastly.deb.debian.org/debian stretch/non-free armhf Packages [71.2 kB]
Get:17 http://cdn-fastly.deb.debian.org/debian stretch/main armhf Packages [9262 kB]
Reading package lists...
E: The method driver /usr/lib/apt/methods/https could not be found.
[ o.k. ] Upgrading base packages [ Armbian ]
Reading package lists...
Building dependency tree...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
[ o.k. ] Installing packages for [ Armbian ]
Reading package lists...
Building dependency tree...
E: Unable to locate package armbian-firmware
[ error ] ERROR in function create_rootfs_cache [ debootstrap-ng.sh:216 ]
[ error ] Installation of Armbian packages failed 
[ o.k. ] Process terminated 
[ error ] ERROR in function unmount_on_exit [ image-helpers.sh:60 ]
[ error ] debootstrap-ng was interrupted 
[ o.k. ] Process terminated 

```

nicopace, 2018-11-28:
@igorpecovnik thanks for the fast response.
Same outcome :(

zador-blood-stained, 2018-11-28:
> `E: The method driver /usr/lib/apt/methods/https could not be found.`

Most likely this is related to all other build issues involving
```
E: Unable to locate package armbian-firmware
```
that were reported several times already, but finally somebody posted the more or less full build output.

I don't have Fiddler at hand, but at least in the browser and with wget `http://apt.armbian.com` redirects to `https://apt.armbian.com` (most likely typical nginx configuration that is suggested for web sites).

```
--2018-11-28 18:06:01--  http://apt.armbian.com/
Resolving apt.armbian.com (apt.armbian.com)... 193.40.101.96, 2001:bb8:4008:ff:5054:ff:fea8:96fc
Connecting to apt.armbian.com (apt.armbian.com)|193.40.101.96|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://apt.armbian.com/ [following]
--2018-11-28 18:06:01--  https://apt.armbian.com/
Connecting to apt.armbian.com (apt.armbian.com)|193.40.101.96|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: '/dev/null'

     0K                                                        71.4M=0s

2018-11-28 18:06:01 (71.4 MB/s) - '/dev/null' saved [452]
```

This means that `apt-transport-https` is required to fetch from the repository, but since we (for some reason) include `armbian-firmware` in the base package list (along with `apt-transport-https` that is required to fetch from it in the first place) obviously people would get this error.

The only thing that most likely "helps" is apt-cacher-ng on the host that should be able to use the https transport by itself, but, obviously, it is not used with Docker.

nicopace, 2018-11-28:
I think this issue is related, though it is not using docker apparently:
https://github.com/armbian/build/issues/1152

igorpecovnik, 2018-11-28:
> redirects to https

Not anymore. But once we put server(s) behind balancer it will break again.

nicopace, 2018-11-28:
Do you want me to try something?

igorpecovnik, 2018-11-28:
Now it should work in any case, try. Later I will revert SSL redirection to see if this is future proof. Have to run now ...

igorpecovnik, 2018-11-29:
I switched back to force SSL redirect and after last fixes this should be resolved. In case not, reopen the issue.