armbian/build #1183: CRYPTROOT feature for using nand-sata-install

ei-ke, 2018-12-10:
Hello team Armbian,
I've just added the CRYPTROOT_PARAMETERS variable in #1182.

Now I'd like to be able to move my encrypted rootfs from my uSD to the attached SATA SSD. I'm using an Odroid HC2, so this step is the next obvious one.

I tried this manually using rsync / chroot but until now without success. I also tried to pick as much as I could from your nand-sata-install script but probably missed some important parts.

Maybe someone here is willing to help me getting started to do this manually as fiddling around with the nand-sata-install is beyond what I can do. I wouldn't consider my self as a coder but rather as someone who can tinker together tiny things.

As I saw that @zciendor and @zador-blood-stained were introducing the CRYPTROOT part into Armbian I think maybe you can also help me with this.

Greetings
Eike

ei-ke, 2018-12-11:
A quick update how I manually solved this for those who want to give it a try.
- Write the Armbian build that includes the encryption to the uSD card and the SSD
- expunge the LUKS partition from the uSD card
- boot from uSD card, set up password and user, shutdown the system as it needs a reboot for resizing the LUKS partition
- system is up and running
- optionally remove the boot partition from the SSD, but doing so before the SSD has been resized will cause an error and you can't boot the system as cryptroot-unlock is running into a timeout

mickmattes, 2020-06-29:
I have installed the Armbian Buster Image with LUKS support on my SD Card. It has 32 GB but somehow only 1.3 GB are used/usable:
[image]

Does anyone have an idea, what I'm doing wrong?
Thanks in advance!

closed 2023-04-29 by rpardini