armbian/build #1184: Change to newer kernel source and Rockchip boot binaries for rockchip64

JpegXguy, 2018-12-13:
The new release `4.4.154-1126-rockchip-ayufan` is out (and also the `4.4.167` upstream patch) which I'm sure you guys would include by yourselves in the near future. Great work on keeping this updated!

My main question is why we aren't using the newest Rockchip binaries for `idbloader.bin`.
In [ayufan's rkbin repository](https://github.com/ayufan-rock64/rkbin/) there are 2 files:
* `rk33/rk3328_ddr_786MHz_v1.13.bin` and
* `rk33/rk3328_miniloader_v2.46.bin`

which are newer than the ones mentioned in `config/sources/rockchip64.conf`. I've been using them as my `idbloader.bin` for a while now with no problems.

martinayotte, 2018-12-13:
Mostly because "time is always the missing ingredient" and lack of resources for tons of boards...
I've done the change locally, currently building, then I will test on my Pine Rock64.
If success, I will do the commit ...

JpegXguy, 2018-12-21:
4.4.167 patched merged in #34bf7c6

closed 2018-12-21 by JpegXguy