armbian/build #1195: Master wrapper script structure.

liftoff-sr, 2018-12-20:
My first impression of the project:

The output (work product) of the project seems very nice.  Using debian packaging is brilliant and that makes it way better than buildroot, and even open-embedded, both of which I have used. 

The build environment is however, not as flexible as it could be, still producing the same quality output.  I am trying to add support for another unsupported board and find the structure of the shell script to be very inflexible and too monolithic.  The inherent know how in the project is quite valuable, that is the micro-procedures and recipes.  The way it is all tied together is like having a TIVO without pause, rewind, or fast forward.  I'd love to help, don't know that it's a short term fix.  Here is what I think would help:

1) find a way to invoked these excellent bash procedures up to a pause point, while still honouring dependencies along that path.

2) make the user configuration file load LAST, so it has a chance of over-riding everything.  

3) boil down to a single configuration file, having 2 configuration files, one of which is under version control is awkward.   Then after editing the one under version control, the big wrapper script complains that you changed your copy of the working tree when you invoke the big script.  That is very awkward.

Is BASEL any good, could this or some other build system be used to conditionally call these micro-bash-procedures based on dependencies already satisfied?  In a normal build environment you can cherry-pick targets and the build system will stop at that point and only after making sure anything needed prior to that is built.  

Again, no complaints about the outputs, or work products of the project.  But the wrapper build script is too monolithic, and although BASH is fine for low level, it is hard to institute dependencies in it.  So we could think about a division of responsibility between two build languages.  Thinking is not doing, and you can think anywhere.

liftoff-sr, 2018-12-20:
The other advantage to having "targets along the way" is that you can have branch points, which are alternative final targets which may rely on identical prior satisfying targets.

And of course you know about that paradigm from package management.

igorpecovnik, 2018-12-20:
Thank you for compliments! We are aware of limitations and just recently we started to rework a part of build script to sort of this mess. At least some off. Proceed on this thread: https://forum.armbian.com/topic/9031-rfc-wip-changes-for-boards-and-features-implementing Welcome.

liftoff-sr, 2018-12-20:
Hi Igor,
I am unable to justify joining the forum at this time.  I appreciate what you have done, what your are trying to do.  I understand the problem space EXTREMELY well.   And I know it is conceptually very difficult to stay motivated as a project leader where there is little reward.  I was project leader of KiCad for about 6-7 years.  At the end, I hated it.  But I got it to critical mass and now I get to enjoy using it.  The new kids probably don't even know me :).  

Be careful how you spend your precious life.  *Always use the highest level tool you can find for the job.*  Somebody once used the phrase "programming on your hands and knees" as a derogatory description of an inefficient programming tool. .  If you have a workshop, you know how important it is to have the right tool for the job.  It's ever more so true in software.  

At this point, after 12 years of experience of open-embedded, 2 years experience with build-root, and gleaning the fine build system of openwrt, I feel like I have gotten nothing but bloody while using this build system.  This is just brutal honesty.  It needs to be decomposed, and long term goals established before being put back together. And yes, shell scripting is not the right tool for the whole thing in my (not so humble) opinion.

There are a couple of projects that take open-embedded as a recipe execution engine, and then try and layer on debian packages, either as compiled recipes or even as binary packages.  This is essentially what I think you need to do.  Your wheel solves a problem, but the tool you are using will not allow you to keep up with the big dogs as they enter the arena.  Just look at this:

https://elinux.org/Shared_Embedded_Linux_Distribution

Then the big dogs come along and say hay, "that there was pretty cool":

https://wiki.linuxfoundation.org/civilinfrastructureplatform/cip-core

Take a week off and learn open-embedded.  You are programming on your hands and knees, and running this script entirely as root is very discrediting to your credibility.  It goofs the version control system repos within the tree and makes the whole thing look almost foolish.  I think this feedback is hopefully helpful.  Life is short, and you do not want to waste it.

All your expertise IS VALUABLE to the industry, and I think you need to reformulate it.  You could actually beat the CIP project to the market.
I truly hope that I have helped you and this project.  In its current form, it is not something that I can help you with other than this advice.
Dick Hollenbeck, CTO SoftPLC Corporation

liftoff-sr, 2018-12-21:
Hi Igor,
One last tip.
Your team's project embodies u-boot and kernel knowledge.  In contrast, this other project [here](https://github.com/ilbers/isar) embodies rootfs knowledge.  I imagine some synergy could be formed between the two projects and you could learn most easily from this other project on how to get your work into bitbake recipes.

I think this project will give you:
a) the easiest learning introduction to a useful bitbake result.
b) does not overlap with your work in a purposeful way, so I can image a union of your project with this one that creates a best in class result.  You can impart your team's script wizardry into the bootloader and kernel portions, which this project does not seriously address (although there is some talk about a bootable image using a tool called WIC.)  It seems not to be the emphasis of the project.

Some background info to accelerate your uptake:
bitbake is the recipe execution engine used in open-embedded.  poky is mostly a solid set of carefully chosen bitbake recipes for a linux distro that layer on top of an open-embedded random recipe library foundation using bitbake's layer priority architecture.  In contrast, this ISAR project seems to go back down to raw bitbake, skip open-embedded altogether, and simply use the recipe engine not to compile source code, as it typical, but to instead grab binaries from debian.  

But in general recipes are the smaller units of configuration management that I was talking about above.  And, they seem to be BASH script fragments embedded into a structured framework that comes complete with dependency management.  I cannot imagine a better path forward for your team.

Hope this helps,
Dick

lanefu, 2019-03-06:
Wisdom appreciated. Closing issue.

closed 2019-03-06 by lanefu