armbian/build #1200: Is it possible to piblish latest rockchip64 kernel

cyberb, 2018-12-26:
There is a latest build with squashfs xz support to fix snap packages. 

Could you upgrade armbiand build?

https://github.com/ayufan-rock64/linux-kernel/releases/tag/4.4.154-1128-rockchip-ayufan

Or do I need to publish a pill request with the tag change?

Thanks.

igorpecovnik, 2018-12-26:
We use our own kernel config: https://github.com/armbian/build/blob/master/config/kernel/linux-rockchip64-default.config#L5668

Changing tag https://github.com/armbian/build/blob/master/config/sources/rockchip64.conf#L46 won't be enough.

Prepare, sent and wait for next nightly kernel rebuild.

cyberb, 2018-12-26:
Not sure how do you bring changes into those files, here is the PR:
https://github.com/armbian/build/pull/1201

igorpecovnik, 2018-12-26:
beta.armbian.com repository is refreshed once per day - switching there and apt get update & upgrade is all you need to do. If you need this in less than 24h you need to invest some of your own time and build kernel on your own. We make this process as simple as possible.

cyberb, 2018-12-26:
I am more interested in the image, https://dl.armbian.com/rock64/
How often do you publish them?

igorpecovnik, 2018-12-26:
Those images are **manually** tested and supported = different level of service. A little more often than this http://cdimage.ubuntu.com/releases/18.04.1/release But again. You can DIY latest image with build tools.

closed 2018-12-26 by igorpecovnik