armbian/build #1202: Banana Pi Pro hang up with legacy kernel on bionic

zerkalica, 2018-12-26:
I use 4.14 image from https://www.armbian.com/banana-pi-pro/
I try to enable vdpau and switch from next to legacy kernel on Armbian Bionic (3.4.113, linux-image-sun7i). And got an hangup at "Starting kernel..." message.  Set verbosity "9" and mkimage doesn't help.

For xenial, bananapipro.bin exists in package linux-xenial-root-bananapi. wget https://apt.armbian.com/dists/xenial/main/Contents-armhf.gz -O-|zcat|grep bananapipro.bin

But this file not found for bionic: wget https://apt.armbian.com/dists/bionic/main/Contents-armhf.gz -O-|zcat|grep bananapipro.bin

Why no bionic desktop images for banana pi pro? Did you drop support for legacy kernel on bionic for old bananas?

I try to ask this on the [forum](https://forum.armbian.com/topic/8810-banana-pi-pro-hang-up-with-legacy-kernel-on-bionic/) first, but nobody answer.

igorpecovnik, 2018-12-26:
> I try to ask this on the forum first, but nobody answer.

Here is a wrong place to ask.

closed 2018-12-26 by igorpecovnik

zerkalica, 2018-12-26:
> Here is a wrong place to ask.

Ok, but where is the right place, where somebody can answer me?

igorpecovnik, 2018-12-26:
Forum is O.K., but warranty for answering never existed. Until project was small support was possible to handle. Now its far from possible. Pressure on us is so big, that waiting time for anything that has normal priority is more than one year. 

Support = "wasting time", "throwing money out of the window". This can be done only up to certain degree, to the degree we can afford. End user donations covers around 10 minutes of support (less than one simple question) each month. 

Shall we adjust support to those levels or rather just leave as is?