armbian/build #1223: tinkerboard wifi/bt (r8723bs) not detected in 5.70

nwf, 2019-01-13:
Reverting to the 5.66 build (i.e. Linux 4.18.18-rockchip) makes it work again, but it's not clear to me if the problem is in the DTB or the kernel itself.  If it is absolutely necessary, I can bisect between b322fec52abe7ba5b42564734410fca3ff79b813 and cadd885d979a24959b879daef4bb6c8f7910b11e, but it's moderately painful to have to babysit a failure on serial console through kernel and dkms reinstall.

On 5.66, dmesg shows
```
[    7.918750] RTL8723BS: module init start
[    7.918755] RTL8723BS: rtl8723bs v4.3.5.5_12290.20140916_BTCOEX20140507-4E40
[    7.918758] RTL8723BS: rtl8723bs BT-Coex version = BTCOEX20140507-4E40
[    8.000397] RTL8723BS: rtw_ndev_init(wlan0)
[    8.001631] RTL8723BS: module init ret =0
```
while on 5.70 the results are similar but the `rtw_ndev_init(wlan0)` line does not appear and no netdev is created.

igorpecovnik, 2019-01-13:
I am just playing with Tinkerboard ATM and BT works (I did connect a mouse) on Bionic Desktop from https://www.armbian.com/tinkerboard Logs: http://ix.io/1ygs

In case you want to proceed resolving, please keep this here: https://forum.armbian.com/topic/3327-asus-tinkerboard/

closed 2019-01-13 by igorpecovnik

nwf, 2019-01-19:
Y...yes?  I believe the log you posted, but you'll note that the kernel identifies as `4.19.14-rockchip` but the version in `linux-image-dev-rockchip_5.70_armhf.deb` (please note the 5.70 in the title of this issue) is `4.20.2-rockchip`.  Fearing I'd done something wrong given the fast close of this bug, I've also built and tested a fresh image using `5.71`, too, which uses the same kernel, and it, too, fails in the same way.

Tonymac32, 2019-01-19:
>  (please note the 5.70 in the title of this issue) is 4.20.2-rockchip

I have done no testing of that kernel yet, I was focused on getting the LTS kernel rolled out properly.  If you change from Dev to Next series kernels you will have 4.19.16, and I am using Wifi with that as we speak.  I will keep an eye out for the wifi issues on 4.20 when I come around to work on it, however that is an extra time sort of activity to make sure I keep pace with the mainline development to prepare for the next LTS.

Tonymac32, 2019-01-19:
Also note, jumping kernels like that may have unintended consequences, be sure to check your device tree, we implemented overlays on 4.19 which did change the root device tree slightly to accommodate in a manner that wasn't insane.

nwf, 2019-01-19:
Thanks for the heads up.  I tripped over this when `apt` automatically upgraded my "production" machine to the next `dev` package and then I went to figure out why.  I'll take your suggestion and switch down to `next` for the moment.

Tonymac32, 2019-01-19:
Dev is by definition unstable, unless you're experimenting I'd avoid Dev kernels.