armbian/build #1225: Building image for rockpi-4b fails at u-boot architecture select

Vonor, 2019-01-16:
Hi,

running the following build command fails.

```
./compile.sh docker KERNEL_ONLY=no KERNEL_CONFIGURE=no BUILD_DESKTOP=no BRANCH=default RELEASE=bionic BOARD=rockpi-4b
```

The abortion happens at U-Boot 2017.09 Configuration Architecture Select (make silentoldconfig) with the following error
```
Console input/output is redirected. run 'make oldconfig' to update configuration

igorpecovnik, 2019-01-16:
This build is under development.

closed 2019-01-16 by igorpecovnik

Vonor, 2019-01-16:
> This board is still under development.

So in a development platform (github) you are clsoing a bug report about something that is still in development? I am sorry, but I may be misunderstanding the concept of a bug report system then.

I would suggest keeping the issue open so people that have knowledge about u-boot and are working on the development of this board can see it.

If I can be of assistance in resolving the issue I would gladly help, but would need some insight of where to look.

igorpecovnik, 2019-01-16:
Read bug report template.

Tonymac32, 2019-01-16:
> I did. There is one point that speaks for going to the forums but two
> points speaking for a bug report here.
>
> Please use the Armbian forum (https://forum.armbian.com/) for:
>
>    - asking for support of new boards or third party hardware (such as
>    wireless and DVB adapters, I2C/SPI sensors, USB touch screens and HDMI
>    displays),
>
> This point speaks for asking on the forums. However, the board is already
> in the development branch of the build script. So not really a "request for
> supporting a new board".
>
> Please use this issue tracker only for the build script related bugs and
> enhancements, such as:
>
>    - discussing the kernel and/or u-boot patches,
>    - discussing changes and enhancements to the image creation process
>
> These two points made me open the request here. Did I miss anything?
>
> Anyway. I would still offer my help in getting this issue fixed, provided
> my knowledge is sufficient for this.

This board is unsupported, the "WIP" tag and the fact you have to go
through a red screen to get to the options to build it fall under the
definition of "unsupported".  Requesting work on the default image should
be done in the forum, hence closing the request.  No one is trying to be
rude, it simply doesn't make sense to have many open bug trackers for a
mountain of csc and wip boards, for each kernel, when there may or may not
be any intention to finish or support that specific thing.  Being
unsupported is not a bug, after all.  If it is supported and does not
build, then it is a bug.