armbian/build #1227: Re-enable XFS filesystem support

rkojedzinszky, 2019-01-16:
Today I've upgraded my orange pi computers where I use Armbian with xfs partitions, and after the upgrade, they failed booting, as xfs support disappeared from linux-sunxi-next (and maybe from others).

igorpecovnik, 2019-01-16:
We need to adjust configurations. This:
https://github.com/armbian/build/blob/78999fc94bdbf2907ae64a2c30437da4c3f1a035/config/kernel/linux-sunxi-next.config
with current
https://github.com/armbian/build/blob/master/config/kernel/linux-sunxi-next.config

rkojedzinszky, 2019-01-16:
Yes, I thought that something like that should be done. That seems simple, isn't it?

rkojedzinszky, 2019-01-16:
Tests shows that only enabling the xfs as a module is fine, initramfs will handle booting. Should I create a pull-request for it?

closed 2019-01-20 by igorpecovnik

hgp9102, 2019-02-07:
We still face the same issue with the unsupported filesystem XFS in version kernel version 4.14.84-sunxi (armbian 5.73)

hgp9102, 2019-02-07:
created new case https://github.com/armbian/build/issues/1269

PR #1239 (2019-01-20): Enabled XFS filesystem support
Fixes #1227 

Enabled XFS in all kernel configurations.
commit 0c69ec0c (2019-01-20): Enabled XFS filesystem support

Fixes #1227