armbian/build #1229: U-Boot for OrangePi Zero Plus2 H3 fails to build since upgrade to 2018.11

hck666, 2019-01-17:
For the past couple of days I have tried to build Armbian for OrangePi Zero Plus2-H3, but even though it did succeed a week ago, since a couple of days I am getting errors from building u-boot.

  DTC     arch/arm/dts/sun8i-h3-orangepi-zero-plus2.dtb
arch/arm/dts/sun8i-h3-orangepi-zero-plus2.dtb: ERROR (phandle_references): /soc/mmc@1c0f000: Reference to non-existent node or label "mmc0_pins_a"

arch/arm/dts/sun8i-h3-orangepi-zero-plus2.dtb: ERROR (phandle_references): /soc/mmc@1c10000: Reference to non-existent node or label "mmc1_pins_a"

```
ERROR: Input tree has errors, aborting (use -f to force output)
scripts/Makefile.lib:310: recipe for target 'arch/arm/dts/sun8i-h3-orangepi-zero-plus2.dtb' failed
make[2]: *** [arch/arm/dts/sun8i-h3-orangepi-zero-plus2.dtb] Error 2
dts/Makefile:33: recipe for target 'arch/arm/dts/sun8i-h3-orangepi-zero-plus2.dtb' failed
make[1]: *** [arch/arm/dts/sun8i-h3-orangepi-zero-plus2.dtb] Error 2
Makefile:930: recipe for target 'dts/dt.dtb' failed
make: *** [dts/dt.dtb] Error 2

```

The problem seems to be in build/patch/u-boot/u-boot-sunxi/add-orangepi-zeroplus2_h3.patch, which was modified 3 days ago. The dts file that the patch generates, refers to mmcX_pins_a, where all other dts files have mmcX_pins (without the _a extension).

I see that that commit changed all other occurences of mmcX_pins_a, except this one. When I did that as well (changed the patch file, removed the two _a's), I can compile for at least KERNEL_ONLY="yes" (have not tried further than that)

martinayotte, 2019-01-17:
Thanks for reporting !
Fixed : https://github.com/armbian/build/commit/f165d351ef16253ed758259e5ec361609927be27

closed 2019-01-17 by martinayotte

hck666, 2019-01-17:
Cheers! :)