armbian/build #1250: linux-mvebu -next and -dev missing USB_NET_PLUSB

cyanidium, 2019-01-28:
USB-to-USB Cable Bridge from Prolific is not recognised on my Helios4 because CONFIG_USB_NET_PLUSB is not set for -next and -dev kernels. Cable is detected and works with -default, where it's a module.

Can you re-enable the module for the newer kernels?

igorpecovnik, 2019-01-28:
You can add this on your own as well. https://www.armbian.com/get-involved/

https://github.com/armbian/build/blob/master/config/kernel/linux-mvebu-next.config
https://github.com/armbian/build/blob/master/config/kernel/linux-mvebu-dev.config

closed 2019-01-28 by igorpecovnik

cyanidium, 2019-01-28:
Sure, let me just download the right version of ubuntu, install it, get all the build packages, clone the repo, make the two line change (via menuconfig for some reason) and submit a PR. While I'm watching ubuntu install, please enjoy some [yaks](https://www.hanselman.com/blog/YakShavingDefinedIllGetThatDoneAsSoonAsIShaveThisYak.aspx).

igorpecovnik, 2019-01-28:
> make the two line change

You can edit on github. It's a 5 minute job, without wasting our time. You need something ...