armbian/build #1256: How to I influence the package names regarding the substring with the kernel version?

Staars, 2019-01-30:
Currently I am trying to build armbian for the Realtek-1295-platform, which works quite okay, but especially for building full OS images, I get a problem with the naming conventions inside the build scripts.
In the snippets below, we see that the kernel version (...4.9.153..) is included in the package name.
Later on the scripts complains, that the name should be "linux-image-rtd1295_5.74_arm64.deb".
This makes it cumbersome to build the full os images, because my only work-around is to first build the kernel package, save and rename it elsewhere and the copy it back to the output folder, after starting the second build (full os).

`dpkg-deb: building package 'linux-firmware-image-4.9.153-rtd1295' in '../linux-firmware-image-4.9.153-rtd1295_5.74_arm64.deb'.
dpkg-deb: building package 'linux-headers-4.9.153-rtd1295' in '../linux-headers-4.9.153-rtd1295_5.74_arm64.deb'.
dpkg-deb: building package 'linux-libc-dev' in '../linux-libc-dev_5.74_arm64.deb'.
dpkg-deb: building package 'linux-image-4.9.153-rtd1295' in '../linux-image-4.9.153-rtd1295_5.74_arm64.deb'.
dpkg-deb: building package 'linux-image-4.9.153-rtd1295-dbg' in '../linux-image-4.9.153-rtd1295-dbg_5.74_arm64.deb'.
dpkg-genchanges: info: binary-only upload (no source code included)
dpkg-deb: building package 'linux-source-4.9.153-default-rtd1295' in '/home/vagrant/armbian/.tmp/linux-source-default-rtd1295_5.74_all.deb'.
dpkg-deb: error: failed to read archive '/home/vagrant/armbian/output/debs/linux-image-rtd1295_5.74_arm64.deb': No such file or directory`

I really tried to educate myself from other sources and I do not have a lot of experience with that, but

igorpecovnik, 2019-01-30:
> is included in the package name

Use packaging patch. Try those two:
https://github.com/armbian/build/blob/ae14e90a304af3c3449ad70506a87db2a353f13c/patch/kernel/odroidxu4-next/packaging-4.x-NEXT-with-postinstall-scripts.patch
https://github.com/armbian/build/blob/ae14e90a304af3c3449ad70506a87db2a353f13c/patch/kernel/odroidxu4-next/bash_to_afterinstall.patch

if not get the one from some 4.14.y kernel.

igorpecovnik, 2019-01-30:
I see you don't have patch directory for this kernel at all. Create:
patch/kernel/rtd1295-default
and put above patches in. At least those two, others are optional.

Staars, 2019-01-31:
I happily admit, that I am even more confused now than before, but it seems to work after some small changes. In my setup I removed the "-next"-substrings from the patch and now the compilation showed no errors.
I will do some tests with new images on the devices and after that close this issue.
Thank You!!

igorpecovnik, 2019-01-31:
We use those packaging patches to deal with naming, which has to be different than stock Debian, which deal with one kernel per architecture. We have one kernel per SoC.

NEXT and DEV are here to divide this further. Default (without prefix) is usually stock kernel, NEXT is usually some mainline branch and DEV master branch. Sometimes this is a bid changed.

Staars, 2019-01-31:
Ahh, now it is getting clearer! 
I never took a closer look at the various patches for the other board families and now I see, that this is a usual procedure.
So, Thanks again! 

(more stuff about my realtek-build will follow in the armbian forum)

closed 2019-01-31 by Staars