armbian/build #1261: Wireguard handshakes

lampra1, 2019-02-03:
With this [commit](https://git.zx2c4.com/WireGuard/commit/?id=c870c7af53f44a37814dfc76ceb8ad88e290fcd8), wireguard uses __kernel_timespec for handshake time.

On Linux cubietruck 4.19.17-sunxi #5.73 SMP Mon Jan 28 21:06:40 CET 2019 armv7l GNU/Linux using the latest wireguard snapshot 0.0.20190123, handshakes are not reported any more.

igorpecovnik, 2019-02-04:
We include wireguard source at build time. If it's broken in any way, you need to proceed with your own solution: disable module and build a new (possible fixed one), use nightly kernel builds which uses latest sources, ...

igorpecovnik, 2019-02-04:
I found a problem - we leave old version if exists. This need to be fixed.

igorpecovnik, 2019-02-06:
Wireguard is now always latest https://github.com/armbian/build/commit/e959294b7ff9a023b3c39dfcb90a56a42a619cab#diff-a3c25b5813cd3787e2da5bdb0e50f437 

Upstream branch can be altered here https://github.com/armbian/build/blob/master/lib/main.sh#L251 if needed.

closed 2019-02-06 by igorpecovnik