armbian/build #1268: Wireguard build fails (Rockchip64)

JpegXguy, 2019-02-06:
``` 
 CC [M]  net/wireguard/noise.o
net/wireguard/noise.c: In function ‘tai64n_now’:
net/wireguard/noise.c:461:16: error: implicit declaration of function ‘ALIGN_DOWN’; did you mean ‘PFN_DOWN’? [-Werror=implicit-function-declaration]
  now.tv_nsec = ALIGN_DOWN(now.tv_nsec,
                ^~~~~~~~~~
                PFN_DOWN
cc1: some warnings being treated as errors
scripts/Makefile.build:277: recipe for target 'net/wireguard/noise.o' failed
make[1]: *** [net/wireguard/noise.o] Error 1
Makefile:1643: recipe for target 'net/wireguard/noise.o' failed
make: *** [net/wireguard/noise.o] Error 2
```

igorpecovnik, 2019-02-07:
Disabling it for this family until patch is created.

closed 2019-02-07 by igorpecovnik

JpegXguy, 2019-02-07:
Does it actually work for you? No build errors? Seems like it;s a warning that is being treated as an error.
Do other families not have this problem?

Adding this [ `#define ALIGN_DOWN(value, align) ((value) & ~((align) - 1))` ] to `net/wireguard/noise.c`  works but I have no Idea how to apply it as a normal patch in the build process. Patches are inserted before Wi-Fi (rtl8812au) and Wireguard. Maybe the additions should come before the (user)patches.

Patch could be something like this, for example.
```
diff --git a/net/wireguard/noise.c b/net/wireguard/noise.c
index 4405125..2e572b2 100644
--- a/net/wireguard/noise.c
+++ b/net/wireguard/noise.c
@@ -17,6 +17,8 @@
 #include <linux/highmem.h>
 #include <crypto/algapi.h>
 
+#define ALIGN_DOWN(value, align) ((value) & ~((align) - 1))
+
 /* This implements Noise_IKpsk2:
  *
  * <- s
```
For the same reason as mentioned above, I cannot use CREATE_PATCHES=yes in order to include is normally.

Also setting WIREGUARD to yes asks me 2 questions, maybe missing config?
```
  IP: WireGuard secure network tunnel (WIREGUARD) [M/n/y/?] (NEW) 
    Debugging checks and verbose messages (WIREGUARD_DEBUG) [N/y/?] (NEW) 
```

igorpecovnik, 2019-02-07:
Adding patches here is not (yet) supported. For example [I am using SED](https://github.com/armbian/build/blob/master/lib/compilation.sh#L293-L294) to fix sources before compilation, but it's also possible to invoke a patch from perhaps patch/misc location?

JpegXguy, 2019-02-07:
Would be cool if it was supported. BTW, there is also the path of waiting until it's fixed upstream, but maybe it doesn't present as an issue for them. Does it build for you without the "patch"?

igorpecovnik, 2019-02-07:
yes, it works. I am already working on implementation.

JpegXguy, 2019-02-07:
I made a PR https://github.com/armbian/build/pull/1272

igorpecovnik, 2019-02-07:
No need, fixed with a patch - the rest is also adjusted. But it would be best to move this entire section under own file to get more control. I also plan to move in AUFS which will probably be a bit more complicated and messy.

JpegXguy, 2019-02-07:
Alright, nice job!

igorpecovnik, 2019-02-11:
Perhaps we have [another problem](https://forum.armbian.com/topic/9602-linux-headers-41920-cubox-make-scripts-fails/?tab=comments#comment-72225) related to Wireguard?

JpegXguy, 2019-02-11:
To be fair, there's a reason it's not in mainline yet. Though this one doesn't seem to be exactly development-related