armbian/build #1275: About the rtl8812au repo

JpegXguy, 2019-02-10:
Do we do this, as suggested? https://github.com/aircrack-ng/rtl8812au

> But for RPI 3 B+ you will need to run those below which builds the ARM64 arch driver:
> ```
> $ sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile
> $ sed -i 's/CONFIG_PLATFORM_ARM64_RPI = n/CONFIG_PLATFORM_ARM64_RPI = y/g' Makefile
> ```

igorpecovnik, 2019-02-10:
It works OOB. I would say we don't touch it.

JpegXguy, 2019-02-10:
It works... kind of. But that's realteks problem , not ours. I'll try to see what it does if I change the options. I really think we should follow the suggested path for **ARM** devices on **Arm**bian. Unless that flag is only for RPi. Time to test.

Seems like there the flags are CONFIG_PLATFORM_ARM_RPI and CONFIG_PLATFORM_ARM64_RPI. Maybe that arm path is better for our boards. Of course I don't know for sure. Time to test!

igorpecovnik, 2019-02-10:
IIRC I tried this kind of obvious flag some time ago and it didn't bring any change. Try.

JpegXguy, 2019-02-10:
I guess we can leave it for now. No need to insert additional complexity to the builds while it works.

closed 2019-02-10 by JpegXguy