armbian/build #1279: In nand-sata-install - blkid hangs on Pine64 LTS Ubuntu_bionic_next

Anton-Latukha, 2019-02-15:
Good day.

If somebody has more info - please add in the comments.

Using Pine64 LTS, flashed Armbian_5.75_Pine64so_Ubuntu_bionic_next_4.19.20 from https://dl.armbian.com/pine64so/Ubuntu_bionic_next.7z (checksum checked) on 2019-02-10 to SD card.

1. There is Pine64 LTS with erased eMMC module.
2. Plugged-in SD card and booted the Pine64 LTS.
3. Login over SSH as root.
4. Run `nand-sata-install`
Observe the full hang of the session.

With Bash debug (`-x`) observed that `blkid` hangs on the system.

Anton-Latukha, 2019-02-15:
`blkid` seems to be pretty known for hangs:
https://bugs.launchpad.net/ubuntu/+source/util-linux/+bug/528073
https://bugzilla.redhat.com/show_bug.cgi?id=1121691
https://bugzilla.redhat.com/show_bug.cgi?id=1591603

And none of the found by me reports was solved, and `blkid` was left unchanged.
In RadHat 1591603 is said something between the lines that it is how it is created by designed and how in RHEV they use a workaround to work with `blkid`.
That is basically why I wrote here and not went to `util-linux`.

I deduced that what I am having can be a similar case to this old issue: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=588375

And indeed - it was. I recreated a `msdos` partition table on eMMC and created a `ext4` partition there.
And after that the `blkid` worked, and `nand-sata-install` ran.

Anton-Latukha, 2019-02-15:
What to do further with this report I let on your decision.

igorpecovnik, 2019-02-17:
Pine64 has many hw design flaws so I suspected hardware failure at first. I apologize for being rude, but when I saw someone opening many issues for the same problems, all at wrong place, I usually get pissed off and close them all. We are too small crew and I am helpless to address those problems especially if a bug comes from upstream. This project is maintained by just a few people, sponsored mainly from our pockets.

If blkid is making troubles, than killing and creating an empty ext4 partition on eMMC would be a possible workaround?

Anton-Latukha, 2019-02-17:
@igorpecovnik Yes, as I wrote above - this report was solved by it.

I would try to make PR in the documentation regarding several things I experienced recently, like this one. But I am too fresh to this project to know.

Don't look at bug reports as something personal, only of course if some management looks at them in a crooked way. Bug reports is the place for people to share info and find solutions, many times someone else finds and posts the solution, makes a PR. Concentrate on the main aspects for you, and do them right - and there would be fewer bug reports regarding your previous work. I never was in the head of a big Open or Free project, and I can not imagine the volume you are working with. In a big part - reporters just need to be sent to the roughly right direction, and they would work themselves.
I try to adopt the: develop -> think -> found no solution -> issue -> connection/direction/info -> dev -> PR -> merge.

I would try to contribute a little, as my current time would allow me, currently, I am pretty economically and time exhausted, but I would carve and fix some of the things on my way.

Anton-Latukha, 2019-02-17:
Here issue was - `blkid` is pretty "hangy" utility. It hanged because first sectors of eMMC was erased (which is a pretty legit usage case in *nix world, to erase just boot/partition table for OS is analogous to erase the whole drive, ask my erased LVM).

I would send docs regarding this, because `blkid` probably hangs in all cases where people do:
```
dd bs=4M if=/dev/null of=/dev/sdb status=progress oflag=sync
```

TRSx80, 2020-12-01:
@Anton-Latukha,

I am struggling to think where I would even put this into our [documentation](https://docs.armbian.com/).  And this even after I have become somewhat familiar with Docs, making a few contributions lately.

Outside the scope of this particular issue, I think we have lots of random little bits of info here and there which are not particular to Armbian and therefore do not really belong in official Docs, however should maybe be noted "somewhere."  And I get that this is the spirit in which you submit this issue.  My personal belief is that we should have something like a wiki for such information.  However there is none officially (yet) and to create one is a larger, project level decision (which has not even been discussed yet, much less made).

In the meantime, I have started to collect lots of little nuggets like this into my own personal notes, and I will also add this one.  Maybe some day that will become the basis of a new wiki.

In the meantime, I think I will close this as "not our bug" and hope that people should be able to find it by searching.

Thank you for your research and contribution!

closed 2020-12-01 by TRSx80

Anton-Latukha, 2020-12-02:
Well, I never send the docs.

Thank you if you mention this for people.

AFAIR, when I looked, and looking currently - `blkid` was/is an unmaintained project, so bugs like these would hang around for people.

Thank you for your contributions to humanity software.