armbian/build #1280: nand-sata-install results in endless loop with rsync

Anton-Latukha, 2019-02-15:
Good day.

If somebody has more info - please add in the comments.

Using Pine64 LTS, flashed Armbian_5.75_Pine64so_Ubuntu_bionic_next_4.19.20 from https://dl.armbian.com/pine64so/Ubuntu_bionic_next.7z (checksum checked) on 2019-02-10 to SD card.

1. There is Pine64 LTS with 16Gb eMMC that has `msdos` partition table, clean primary partition type `ext4` with `ext4` filesystem on it.

2. Plugged-in SD card and booted the Pine64 LTS.

3. Updating the distribution on SD card:
```
apt update
apt upgrade
reboot
```

4. Run `bash -x /usr/sbin/nand-sata-install`

If we run without `-x` we would observe pseudographics UI, and during copying of data it would report ETA 2 minutes and hang on 6%.

With `-x` endless loop can be observed.

https://github.com/armbian/build/blob/c53f5611b30d65276ff6663d2789141c14fa260c/packages/bsp/common/usr/sbin/nand-sata-install#L167-L193

With:
```shell
prev_progress=22
rsync_progress=22
```

Looking at the output and reading the code it is obvious that:
https://github.com/armbian/build/blob/c53f5611b30d65276ff6663d2789141c14fa260c/packages/bsp/common/usr/sbin/nand-sata-install#L179

Indefinitely recieves status `no`.

That means that:
https://github.com/armbian/build/blob/c53f5611b30d65276ff6663d2789141c14fa260c/packages/bsp/common/usr/sbin/nand-sata-install#L154-L160
never results into `echo ${PIPESTATUS[0]} >"${nsi_conn_done}"`

And indeed:
`/tmp/nand-sata-install/progress` consists of 1-21 numbers and then 22 added indefinitely.
`/tmp/nand-sata-install/done` consists of `no`.

And indeed - separately logging into the system it can be observed that there are:
```shell
root@pine64so:/tmp/nand-sata-install# ps -ef|grep rsync
root      1665  1664  0 17:20 tty1     00:00:05 rsync -avrltD --delete --exclude-from=/usr/lib/nand-sata-install/exclude.txt / /mnt/nand-sata-install.Tzl1g6/rootfs
root      1668  1665  0 17:20 tty1     00:00:00 rsync -avrltD --delete --exclude-from=/usr/lib/nand-sata-install/exclude.txt / /mnt/nand-sata-install.Tzl1g6/rootfs
root      1669  1668  0 17:20 tty1     00:00:07 rsync -avrltD --delete --exclude-from=/usr/lib/nand-sata-install/exclude.txt / /mnt/nand-sata-install.Tzl1g6/rootfs
root     25632 20210  0 18:35 pts/0    00:00:00 grep --color=auto rsync
```
RSyncs that that never end. It loops for a couple of hours already.

In last month there was a fresh merge (#1245) of full rewrite to copying procedure that we use here. I can not directly notify a person because commits marked as non-exising on GitHub person guillem.

igorpecovnik, 2019-02-16:
https://github.com/armbian/config/issues/44

closed 2019-02-16 by igorpecovnik

dedalodaelus, 2019-05-26:
Dear Anton-Latukha,
I modified that part of the code and just saw it. The issue continued?
I don't know why the rsync is stuck on the 22 on your board. By the way I understand you correctly with old way you will have the same problem.
I used this copying method on more than 5000 units with no problems (except the hardware problems)

Anton-Latukha, 2019-05-30:
Ok. Thanks.

Maybe it was hardware, but not looked like it, I made couple of tests then. But with hardware - you never know. I was in the process, ran into this, so I switched to other tasks.
Hope I was as tactful as possible.
The good thing is this resulted in couple of patches sent from my side.

I would try again somewhere in the future, probably on the same board/s.