armbian/build #1286: Adding kernel support for DVB-T2 USB device MyGica T230C2

pevot, 2019-02-19:
The next kernel contains support for DVB-T2 USB devices MyGica T230 and DVB-T2 USB device MyGica T230C but not for the latest revision MyGica T230C2
Would it be possible to add kernel support for this device into the next and dev kernels before it will be added to the official kernel?
Following two patches are necessary to do this:
https://patchwork.kernel.org/patch/10395967/
https://patchwork.kernel.org/patch/10395963/
I've tested it as userpatches in my build for OrangePi One and Cubieboard2 and it works fine.

For the correct functionality the following firmware files are necessary:
dvb-demod-si2168-d60-01.fw
dvb-tuner-si2141-a10-01.fw
If it is possible you can add them too, but they can be easy copied to the system by the user.

igorpecovnik, 2019-02-20:
> Would it be possible to add kernel support for this device into the next and dev kernels before it will be added to the official kernel?

Sure, you can do most of the tasks. 

> I've tested it as userpatches in my build for OrangePi One and Cubieboard2 and it works fine.

Prepare PR https://www.armbian.com/get-involved containing those two patches.

> For the correct functionality the following firmware files are necessary:

Send PR here:
https://github.com/armbian/firmware

pevot, 2019-02-20:
OK I will prepare the PRs.

pevot, 2019-02-20:
PRs submitted. I hope I did everything right.

closed 2019-02-25 by pevot