armbian/build #1288: Build FIRMWARE ?

150balbes, 2019-02-20:
This is just me, during build, each time you start the process of downloading the entire GIT with the firmware ?

[ .... ] Installing [ armbian-config_5.76_all.deb ]
[ o.k. ] Merging and packaging linux firmware [ @host ]
[ o.k. ] Checking git sources [ armbian-firmware-git master ]
[ .... ] Up to date 
[ o.k. ] Merging and packaging linux firmware [ @host ]
[ o.k. ] Checking git sources [ armbian-firmware-git master ]
[ .... ] Up to date 
[ o.k. ] Checking git sources [ armbian-firmware-full/lib/firmware master ]
[ .... ] Creating local copy 
[ .... ] Fetching updates 
remote: Counting objects: 1940, done.
remote: Compressing objects: 100% (1338/1338), done.
remote: Total 1940 (delta 667), reused 1586 (delta 498)
Receiving objects: 100% (1940/1940), 127.13 MiB | 1.17 MiB/s, done.
Resolving deltas: 100% (667/667), done.
From https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware
 * branch            master     -> FETCH_HEAD
 * [new branch]      master     -> origin/master
[ .... ] Checking out 
[ .... ] Installing [ armbian-firmware_5.76_all.deb ]

igorpecovnik, 2019-02-20:
Yes, this part is not done properly and will be ditched completely when "tvboxes" branch become default choice.

mzhboy, 2019-05-15:
problem solved  
fixed in 50dcb63

closed 2019-05-15 by igorpecovnik