armbian/build #1292: Missing Build Dependency: cpio

CurtisLeeBolin, 2019-02-22:
[ .... ] Creating packages
/bin/bash ./scripts/package/mkdebian
dpkg-buildpackage -r"fakeroot -u" -a$(cat debian/arch) -b -nc -uc
dpkg-buildpackage: warning: using a gain-root-command while being root
dpkg-buildpackage: info: source package linux-4.19.24-meson64
dpkg-buildpackage: info: source version 5.76
dpkg-buildpackage: info: source distribution bionic
dpkg-buildpackage: info: source changed by Igor Pecovnik <igor.pecovnik@****l.com>
dpkg-buildpackage: info: host architecture arm64
dpkg-buildpackage: warning: debian/rules is not executable; fixing that
 dpkg-source --before-build linux-4.19.y
dpkg-checkbuilddeps: error: Unmet build dependencies: cpio
dpkg-buildpackage: warning: build dependencies/conflicts unsatisfied; aborting
dpkg-buildpackage: warning: (Use -d flag to override.)
scripts/package/Makefile:79: recipe for target 'bindeb-pkg' failed
make[1]: *** [bindeb-pkg] Error 3
Makefile:1360: recipe for target 'bindeb-pkg' failed
make: *** [bindeb-pkg] Error 2
dpkg-deb: building package 'linux-source-4.19.24-next-meson64' in '/buildroot/build/.tmp/linux-source-next-meson64_5.76_all.deb'.
mv: missing destination file operand after '/buildroot/build/output/debs/'
Try 'mv --help' for more information.
[ error ] ERROR in function compile_kernel [ compilation.sh:429 ]
[ error ] Failed moving kernel DEBs
[ o.k. ] Process terminated

After installing package `cpio`, the build completed.

Tonymac32, 2019-02-22:
I just built this exact image less than 24 hours ago, with no issues.  Are
you using a Bionic build host/container as required?

CurtisLeeBolin, 2019-02-25:
Canonical's Official Ubuntu:18.04 docker image.  I tried again with a fresh container, same dependency missing.

igorpecovnik, 2019-02-25:
Why don't you use officially recommended image instead?

CurtisLeeBolin, 2019-02-25:
@igorpecovnik, I am using the officially recommended image.

igorpecovnik, 2019-02-25:
"Supported build environment is Ubuntu Bionic 18.04 x64 (**minimal iso image**)." ... which is something completely different than Ubuntu Bionic **Docker image** which we don't recommend.

Do you perhaps build under Docker?

lanefu, 2019-03-06:
closing due to commit 3a89b7 addressing issue

closed 2019-03-06 by lanefu