armbian/build #1306: Rockchip build kernel error

Qedr1, 2019-03-10:
I have error kernel in build script for next single board:
NanoPiFire3
NanoPiM4
NanoPiZero
RockPro64

[ error ] ERROR in function compile_kernel [ compilation.sh:387 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated
Log files in   output/debug directory  does not contained any errros

Problem kernel
rk3399
rockchip-default
rk3399-default
rockchip64-default

Last update for build scripts at 2019/03/10

igorpecovnik, 2019-03-10:
> NanoPiM4

... was changed upstream and patches need to be adjusted. The rest must work.

Qedr1, 2019-03-10:
ok, thx for your great job!

igorpecovnik, 2019-03-10:
https://github.com/armbian/build/commit/d7b601b63fbfac6d78c424e344fc95952a8b6f1d

closed 2019-03-10 by igorpecovnik