armbian/build #1311: OrangePi R1 build driver error

Qedr1, 2019-03-14:
In file included from drivers/net/wireless/realtek/rtl8723cs/include/drv_types.h:35:0,
                 from drivers/net/wireless/realtek/rtl8723cs/hal/phydm/phydm_types.h:172,
                 from drivers/net/wireless/realtek/rtl8723cs/hal/phydm/phydm_precomp.h:24,
                 from drivers/net/wireless/realtek/rtl8723cs/hal/phydm/phydm_beamforming.c:2:
drivers/net/wireless/realtek/rtl8723cs/include/wifi.h:1009:0: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40

In file included from ./include/net/cfg80211.h:24:0,
                 from drivers/net/wireless/realtek/rtl8723cs/include/osdep_service_linux.h:92,
                 from drivers/net/wireless/realtek/rtl8723cs/include/osdep_service.h:41,
                 from drivers/net/wireless/realtek/rtl8723cs/include/drv_types.h:32,
                 from drivers/net/wireless/realtek/rtl8723cs/hal/phydm/phydm_types.h:172,
                 from drivers/net/wireless/realtek/rtl8723cs/hal/phydm/phydm_precomp.h:24,
                 from drivers/net/wireless/realtek/rtl8723cs/hal/phydm/phydm_beamforming.c:2:
./include/linux/ieee80211.h:1442:0: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100

This error repeats infinitely in build script and compile kernel does not continiune

igorpecovnik, 2019-03-14:
Which kernel? Did you change anything? Where are logs?

Qedr1, 2019-03-14:
Kernel 4.19
Do not change anything and do not aplay patch
FULL kernel build logs consist of this error

igorpecovnik, 2019-03-14:
But this is not error. Does compilation stops?

lanefu, 2019-03-15:
So i just saw that same warning repeat while building for Opi Lite..... but it was still continuing to compile modules and after a few minutes the warning subsided.    The build completed successfully.

martinayotte, 2019-03-15:
I saw that too, but it is warning about redefinition, the kernel should be the good definition.

igorpecovnik, 2019-03-15:
Worse case, this wireless driver is broken  ... which is anyway out of our path.

closed 2019-03-15 by igorpecovnik