armbian/build #1318: /boot too small w/ nand-sata-install script

lanefu, 2019-03-20:
When using BTRFS, nand-sata-install makes a dedicated /boot ext4 boot partition (for obvious reasons) ...but it seemed to be about 50 megs.. which left me with insufficient space to install new test kernels etc with `dpkg -i` 

https://github.com/armbian/build/blob/master/packages/bsp/common/usr/sbin/nand-sata-install#L465

anyway `131071 sectors * 512 bytes` == not enough space.

I'd prefer to double it.

igorpecovnik, 2019-03-20:
> I'd prefer to double it.

Go.

ThomasKaiser, 2019-03-21:
> I'd prefer to double it

Why?

It's static partitioning, something from the past. The space allocated is lost for anything else. It's 'only' 64GB but still lost for anybody else on this planet wanting to use btrfs on eMMC (since I doubt anyone else will then test through a bunch of different kernels).

Why do you want to adjust `nand-sata-install` as part of Armbian and not your local copy? And why don't you simply choose ext4 the next time you run `nand-sata-install` to avoid the issue at all on a developer install testing through a bunch of kernels?

lanefu, 2019-03-21:
> It's static partitioning, something from the past. The space allocated is lost for anything else. It's 'only' 64GB but still lost for anybody else on this planet wanting to use btrfs on eMMC (since I doubt anyone else will then test through a bunch of different kernels).

@ThomasKaiser 

Not trying to be pedantic, but assuming you meant `64MB`

Isn't upgrading kernels something that other people using BTRFS on eMMC something they'd want to do?  Seems like a reasonable use case.   Maybe they'll reclaim the space wasted to the boot partiton by using BTRFS compression on their root volume 😄 

> Why do you want to adjust nand-sata-install as part of Armbian and not your local copy? 

I saw this as a bug that others would experience.  I was approaching this as spending time improving the build platform and related accessories rather than adding features.  You should be pleased.

Do you have any other counter proposals other than what you've presented already?   If you like, I can run the numbers and try to tighten the space expansion up a bit... Increasing the space by 50% rather than 200% might satisfy the intention.

>  And why don't you simply choose ext4 the next time you run nand-sata-install to avoid the issue at all on a developer install testing through a bunch of kernels?

Because I want to dude.  I'm POCing a le potato as a desktop, and I want BTRFS.

ThomasKaiser, 2019-03-21:
> Isn't upgrading kernels something that other people using BTRFS on eMMC something they'd want to do?

Shouldn't 'upgrading' kernels remove the unneeded left-overs from before?

lanefu, 2019-03-21:
> Shouldn't 'upgrading' kernels remove the unneeded left-overs from before?

Ideally as a clean-up step upon completion.  In my case I was having to delete the old before i installed the new one in order to have sufficient space.

ThomasKaiser, 2019-03-21:
Well, [after increasing the `/boot` partition size](https://github.com/armbian/build/commit/d78d1c43d734404184cf4efe8f32aaa9ca22f2ef) and with a few thousand installs using btrfs (all the OMV images except those for RPi) I've never heard of this partition being still too small (users upgrading and **not** using `dpkg -I`).

And please... if a change is made to `nand-sata-install` the same adjustment with exactly the same logic needs to be made when Armbian images are created and use btrfs.

lanefu, 2019-03-21:
>  I've never heard of this partition being still too small (users upgrading and not using dpkg -i).
Okay that's good to know...  So under normal behavior through an `apt upgrade` from a repo, this edge case doesn't occur... And I suppose that adds up, I realized my Opi2e boards used btrfs and updated okay.   I suppose I could script a more orchestrated process for using dpkg and document.

So given that it's 64 megs.. is your opposition to this in principle or a matter of practicality?

> And please... if a change is made to nand-sata-install the same adjustment with exactly the same logic needs to be made when Armbian images are created and use btrfs.

Good to know.. even if i don't make a change i'll go ahead and put some comments in that script to document the relationship for the future.

ThomasKaiser, 2019-03-22:
> is your opposition to this in principle or a matter of practicality?

I've no idea which problem you try to solve and why you focus on `nand-sata-install` only. If you change something here and not also `$BOOTSIZE` in `debootstrap-ng.sh` at the same time you simply created another support issue ('can not reproduce') once the 64MiB become really too small for kernel upgrades.

When I adjusted this stuff the last time I thought about honoring `$BOOTSIZE` in `nand-sata-install` somehow but gave up since `nand-sata-install` is a mess anyway and still fiddling around with this broken script is insane. Consensus was that a rewrite from scratch is needed but as usual the focus is on fancy stuff all the time.

lanefu, 2019-03-22:
> I've no idea which problem you try to solve and why you focus on nand-sata-install only. If you change something here and not also $BOOTSIZE in debootstrap-ng.sh

I was trying to solve the issue I hit with `dpkg -i`  I wasn't aware of the overlap in `debootstrap-ng.sh`

> at the same time you simply created another support issue ('can not reproduce')

I created an issue as a place holder for the work I intended to do rather than just commiting code.  You'll see that I assigned it to myself.  It  created this dialog where you were able to highlight some gaps in my approach and bring up additional concerns.  Don't give me shit for creating an issue in order to track my work.  

> since nand-sata-install is a mess anyway and still fiddling around with this broken script is insane. Consensus was that a rewrite from scratch is needed

Okay.  Is your ask that I refactor the code to honor $BOOTSIZE as a long-term solution rather than increasing the static coded values?  I can work with that.

> but as usual the focus is on fancy stuff all the time.

This whole issue is literally focusing on stability rather than enhancement.  Granted the original scope was small, but the spirit of improvement is present.    Please don't cite your general grievance that "nobody cares about technical debt" with me again.  I do care.

ThomasKaiser, 2019-03-22:
> Is your ask that I refactor the code to honor $BOOTSIZE as a long-term solution rather than increasing the static coded values?

Neither nor, touching this unmaintainable mess called `nand-sata-install` is wrong since years. Anyway, I don't care what will be done... too tired to babble about the same annoying stuff again and again...

5kft, 2019-08-16:
Hi @lanefu - apologies for dusting off an old thread, but I just hit this same problem in trying to give 5.3.0-rc3 a try.  I recently switched from a btrfs root to f2fs, but the problem is the same - the boot partition is just too small (58MB after formatting):

    Setting up linux-dtb-dev-sunxi64 (5.93) ...
    Setting up linux-headers-dev-sunxi64 (5.93) ...
    Compiling headers - please wait ...
    Setting up linux-image-dev-sunxi64 (5.93) ...
    update-initramfs: Generating /boot/initrd.img-5.3.0-rc3-sunxi64
    
    gzip: stdout: No space left on device
    E: mkinitramfs failure cpio 141 gzip 1
    update-initramfs: failed for /boot/initrd.img-5.3.0-rc3-sunxi64 with 1.
    run-parts: /etc/kernel/postinst.d/initramfs-tools exited with return code 1

/boot is now out of space:

    root@nano:~# df -h /dev/mmcblk2p1
    Filesystem      Size  Used Avail Use% Mounted on
    /dev/mmcblk2p1   58M   53M  1.4M  98% /boot
    root@nano:~#

...and my device is now unbootable without manual cleanup.  Granted this isn't that big of a deal, but it is unfortunate for an otherwise solid distro such as Armbian...it seems like a small matter to simply increase the partition size a bit (honoring $BOOTSIZE or updating the existing hard-coded values or whatever) so that users avoid unwittingly hit this sort of problem in the future.  Where does this stand at this point?  Should I take a look at this?

lanefu, 2019-08-16:
@5kft 

Sorry you ran into the issue.. I had gotten a bit frustrated at the time and walked away from it.... 

Feel free to take a look.. I think if we follow's @ThomasKaiser 's guidance and hit all the touch points he mentioned in the code we will be in good shape

5kft, 2019-08-16:
Hi @lanefu, I took a quick look and I see where @ThomasKaiser is coming from - this isn't a typical use case for most people.

Ideally the two should be tied.  But while one can specify a bootloader partition size for the image build, one cannot for nand-sata-install (with a default fallback to 64MB) - it's hard-coded.

The thing is that if someone makes (or downloads) an image, then they have an ext4 root and there is no bootloader size as it's a single partition.  This would argue for simply adding an option (perhaps hidden) to nand-sata-install to allow one to specify a size for two-partition targets (i.e., btrfs and f2fs).

(Although I have to say it's tempting to just add 32MB to both debootstrap and nand-sata-install and call it a day as this is a tiny enough increase, and I'm hazarding a guess that most people use ext4 anyway... :-))

lanefu, 2019-08-17:
> (Although I have to say it's tempting to just add 32MB to both debootstrap and nand-sata-install and call it a day as this is a tiny enough increase, and I'm hazarding a guess that most people use ext4 anyway... :-))

I think that's the solution we probably need for now.

5kft, 2019-08-17:
Agreed...changes commited with https://github.com/armbian/build/commit/7a2bf78e10d14b6db92b9e4c591d1cf39b2343cf

lanefu, 2019-08-17:
resolved 7a2bf78

lanefu, 2019-08-17:
thanks @5kft

closed 2019-08-17 by lanefu