armbian/build #1330: linux-headers-4.19.20-sunxi64 cannot make scripts

shaddow501, 2019-04-04:
Hello

I cannot "make scripts" on Orange PI zero plus:

root@orangepizeroplus:src/linux-headers-4.19.20-sunxi64 # make scripts
scripts/kconfig/conf  --syncconfig Kconfig
net/Kconfig:89: can't open file "net/wireguard/Kconfig"
scripts/kconfig/Makefile:69: recipe for target 'syncconfig' failed
make[2]: *** [syncconfig] Error 1
Makefile:539: recipe for target 'syncconfig' failed
make[1]: *** [syncconfig] Error 2
Makefile:635: recipe for target 'include/config/auto.conf' failed
make: *** [include/config/auto.conf] Error 2

from older kernels I see that there was a fix, than I do not understand what is the problem now...
what should I do?

lanefu, 2019-04-05:
can you install the kernel-sources package via armbian-config and try?

shaddow501, 2019-04-05:
I did,
same errors:

root@orangepizeroplus:src/linux-headers-4.19.20-sunxi64 # make scripts
scripts/kconfig/conf  --syncconfig Kconfig
net/Kconfig:89: can't open file "net/wireguard/Kconfig"
scripts/kconfig/Makefile:69: recipe for target 'syncconfig' failed
make[2]: *** [syncconfig] Error 1
Makefile:539: recipe for target 'syncconfig' failed
make[1]: *** [syncconfig] Error 2
Makefile:635: recipe for target 'include/config/auto.conf' failed
make: *** [include/config/auto.conf] Error 2

andriej, 2019-04-07:
> if I exclude the - #source "net/Kconfig"
where you do that exactly?

igorpecovnik, 2019-04-08:
apt update and apt upgrade, then try again. It should work now.

closed 2019-04-08 by igorpecovnik