armbian/build #1333: Add USBIP as a module / odroid-xu4

andriej, 2019-04-08:
Please consider adding usbip as a module for default config.
Currently odroid-hc1 has only one USB port and I'd like to connect some devices via USB-IP standard package. 
Unfortunately, as most of the distros have it enabled as module by default (even raspbian with older versions), it didn't make to current kernel of odroid armbian.

Module name: `vhci-hcd`, it's in drives -> usb support -> ip usb

igorpecovnik, 2019-04-08:
Fastest way is if you send a PR to kernel configuration: https://github.com/armbian/build/tree/master/config/kernel

The one(s) with odroidxu4 in the name.

andriej, 2019-04-08:
Opened #1335 - how to proceed further when it gets accepted?
I just can't build it myself somehow (lack of knowledge), would just like to use it.

andriej, 2019-04-08:
@igorpecovnik any chance for this module to become precompiled in apt's sources now?
I have:
`# cat /proc/version                                                                                                                                                                             
Linux version 4.14.94-odroidxu4 (root@armbian.com) (gcc version 6.4.1 20171012 (Linaro GCC 6.4-2017.11)) #14 SMP PREEMPT Fri Feb 8 11:14:13 CET 2019`

The best would be to grab it as .deb

igorpecovnik, 2019-04-08:
> in apt's sources now?

https://forum.armbian.com/topic/5054-nightly-builds/

closed 2019-04-08 by igorpecovnik

lededev, 2021-04-17:
Why not add this settings as kernel global settings, my device RK3288 also want this kernel module.

EvilOlaf, 2021-04-17:
> 
> 
> Why not add this settings as kernel global settings, my device RK3288 also want this kernel module.

Provide PR

lededev, 2021-04-18:
@EvilOlaf  RP https://github.com/armbian/build/pull/2781 Open