armbian/build #1342: Why is vm.swappiness set to 100?

ryanlederman, 2019-04-11:
As I understand it, the `vm.swappiness` setting in `sysctl.conf` controls the tendency of the OS to save to the swap file. The higher the value (0-100), the more likely swapping will occur.

Since my devices are all run on SSD storage, I wouldn't think that swapping _as much as possible_ would be a very good idea&mdash;in theory this will reduce the lifetime of the SSD card.

I found the following in `lib/makeboarddeb.sh`:

```sh
grep -q vm.swappiness /etc/sysctl.conf
case \$? in
  0)
	sed -i 's/vm\.swappiness.*/vm.swappiness=100/' /etc/sysctl.conf
  ;;
  *)
	echo vm.swappiness=100 >>/etc/sysctl.conf
  ;;
esac
```
Can you please explain the decision to always set this value to 100? Cheers.

lanefu, 2019-04-12:
when using zram  based swap you want to encourage the eagerness to swap to take advantage of memory compression.

ThomasKaiser, 2019-04-12:
> The higher the value (0-100), the more likely swapping will occur.

No, it's more complicated than that: https://forum.armbian.com/topic/5565-zram-vs-swap/?page=3

But with Armbian only zram based swap is used and as such `vm.swappiness=100` is the only reasonable choice. Some people have the **feeling** that this is 'too much' but that's usually based on this:
[image]

ryanlederman, 2019-04-22:
@lanefu @ThomasKaiser Understood. Thanks for your replies.

closed 2019-04-22 by ryanlederman