armbian/build #1343: Checking git sources [ armbian-firmware-full/lib/firmware master ] 4kb

TelDragon, 2019-04-12:
Please use the "Preview" tab above to view this message if you are seeing this in the new issue text box.

Please note that this is a **build script** issue tracker.
Issues with provided images should go to the Armbian forum.
Issues with the `armbian-config` application should go to its [repository](https://github.com/armbian/config).
Issues with Device Tree overlays should go to the [dedicated forum thread](https://forum.armbian.com/topic/3787-testers-wanted-sunxi-device-tree-overlays/).

Please use the Armbian forum (https://forum.armbian.com/) for:

 - reporting boot failures (please read [the documentation](https://docs.armbian.com/User-Guide_Getting-Started/) first!),
 - asking for help with Armbian if it is **not** related to building the kernel and/or OS image (please use forum search and Google search first!),
 - reporting kernel crashes or similar issues,
 - asking for support of new boards or third party hardware (such as wireless and DVB adapters, I2C/SPI sensors, USB touch screens and HDMI displays),
 - etc.

Please use this issue tracker only for the build script related bugs and enhancements, such as:

 - requesting changes to the kernel configuration,
 - requesting changes to the default package lists,
 - discussing the kernel and/or u-boot patches,
 - discussing Armbian specific customizations made at image build time,
 - discussing changes and enhancements to the image creation process

Please note that nightly images and dev kernel branches are provided with minimal support, and only build failures with default (provided) configuration can be considered an issue that needs to be fixed.

Please remove the text above before opening an issue.

TelDragon, 2019-04-12:
Is this option？

IGNORE_UPDATES (yes|no): skip downloading and updating sources

igorpecovnik, 2019-04-12:
This part is already completely refactored in "tvboxes" branch but not yet ready for merge.

TelDragon, 2019-04-12:
@igorpecovnik 
Ok, thank you very much for your reply.
Because this problem has been bothering me for a long time, it will waste a long time each time an image is generated.

Do you have a plan for when to merge?

igorpecovnik, 2019-04-12:
> Do you have a plan for when to merge?

The plan was 2/2019 but since its not finished and resources are low. We can't hire help to finish this or that.

> has been bothering me for a long time

Understand. Build script is not ideal in some parts, but development is already going way faster than we actually can afford.

closed 2019-04-19 by TelDragon