armbian/build #1348: Enhancement: Windows Subsystem for Linux Support

AreaScout, 2019-04-15:
## Enhancement report

Enable building on WSL under Windows 10, only that line has to be removed

```
diff --git a/lib/general.sh b/lib/general.sh
index 138fb1e4..c8997712 100644
--- a/lib/general.sh
+++ b/lib/general.sh
@@ -564,9 +564,9 @@ prepare_host()
                fi
        fi

-       if grep -qE "(Microsoft|WSL)" /proc/version; then
-               exit_with_error "Windows subsystem for Linux is not a supported build environment"
-       fi
+       #if grep -qE "(Microsoft|WSL)" /proc/version; then
+       #       exit_with_error "Windows subsystem for Linux is not a supported build environment"
+       #fi

        grep -q i386 <(dpkg --print-foreign-architectures) || dpkg --add-architecture i386
        if systemd-detect-virt -q -c; then
```

and before compile execution of this command has to be issued:

`sudo update-alternatives --set fakeroot /usr/bin/fakeroot-tcp`

### How has this been tested

At least with the `KERNEL_ONLY=yes` switch it's working (that's all I need), I don't know if image building works but I did cross compile other images on WSL already successfully, so it could be possible.

igorpecovnik, 2019-04-15:
Building images is more complicated and it will probably fail. We keep building environment limited also for support reasons. We already struggle to maintain as is.

AreaScout, 2019-04-16:
Yea, I understand that :) Thank you for your fast reply 👍

closed 2019-04-16 by AreaScout

kurai021, 2019-05-03:
I want to test in WSL but I can't find where to put `sudo update-alternatives --set fakeroot /usr/bin/fakeroot-tcp`