armbian/build #1350: Excuse me linux-firmware.git  Is it necessary?

TelDragon, 2019-04-18:
```
[ o.k. ] Checking git sources [ armbian-firmware-full/lib/firmware master ]
[ .... ] Creating local copy 
[ .... ] Fetching updates 
remote: Counting objects: 1960, done.
remote: Compressing objects: 100% (1394/1394), done.
error: RPC failed; curl 56 GnuTLS recv error (-9): A TLS packet with unexpected length was received.
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
```
I compiled it is orangepi-r1

TelDragon, 2019-04-18:
Of course, I will return after the image is generated.
~/build/cache/sources/armbian-firmware-full/lib/firmware
Found that there is content in the current folder and .git/ Missed

TelDragon, 2019-04-18:
I tried to delete the .git/ hidden file after the host with other speed blocks was cloned from the source library. ~/build/cache/sources/armbian-firmware-full/lib/firmware   Regenerating the image is still wrong.

EvilOlaf, 2019-04-18:
Working for me. Also OrangePi-r1
```
[...]
[ o.k. ] Checking git sources [ armbian-firmware-git master ]
[ .... ] Up to date
[ o.k. ] Merging and packaging linux firmware [ @host ]
[ o.k. ] Checking git sources [ armbian-firmware-git master ]
[ .... ] Up to date
[ o.k. ] Checking git sources [ armbian-firmware-full/lib/firmware master ]
[ .... ] Creating local copy
[ .... ] Fetching updates
remote: Counting objects: 1960, done.
remote: Compressing objects: 100% (1368/1368), done.
remote: Total 1960 (delta 679), reused 1559 (delta 487)
Receiving objects: 100% (1960/1960), 129.40 MiB | 9.13 MiB/s, done.
Resolving deltas: 100% (679/679), done.
From https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware
 * branch            master     -> FETCH_HEAD
 * [new branch]      master     -> origin/master
[ .... ] Checking out
[ .... ] Installing [ armbian-firmware_5.79_all.deb ]
[ o.k. ] Building deb [ armbian-tools ]
[ o.k. ] ... downloading sources [ BT utils ]
[ o.k. ] ... compiling [ bluetooth utils ]
[ .... ] Installing [ armbian-tools-stretch_5.79_armhf.deb ]
[ .... ] Installing additional packages [ EXTERNAL_NEW ]
Get:1 http://security.debian.org stretch/updates InRelease [94.3 kB]
[...]
[ o.k. ] Writing U-boot bootloader [ /dev/loop2 ]
[ o.k. ] Done building [ /home/build/build/output/images/Armbian_5.79_Orangepi-r1_Debian_stretch_next_4.19.35.img ]
[ o.k. ] Runtime [ 10 min ]
[ o.k. ] Repeat Build Options [ BOARD=orangepi-r1 BRANCH=next RELEASE=stretch BUILD_DESKTOP=no KERNEL_ONLY=no KERNEL_CONFIGURE=no ]
```

Try starting over with a fresh copy of the build environment.

igorpecovnik, 2019-04-18:
Could be network related trouble.

TelDragon, 2019-04-19:
@EvilOlaf 
Hello, may I ask why my display is different from yours?
My not printing
From https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware

EvilOlaf, 2019-04-19:
I have no idea.
I see you are Chinese user, do you have issues with government firewalling or similar? Just guessing...

TelDragon, 2019-04-22:
@EvilOlaf 
It is true that there are some reasons for this.

TelDragon, 2019-04-23:
@igorpecovnik 
Sorry, bother, I saw the update history of your "firmware.sh" script. 2017/10/07 
Already added your git source。 
and "https://github.com/armbian/firmware" warehouse Also keep updating。
work as “https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware”Cannot clone properly
Firmware is also available. ？

igorpecovnik, 2019-04-23:
Our firmware repository keep only minimum of files, those matched with on-board wireless adapters. For main Linux firmware repository I don't know if there is any official Github mirrors. That could probably solve problem you have.

closed 2019-04-28 by TelDragon