armbian/build #1352: rfkill is not used correctly when setup bluetooth with ap6212

xiongyihui, 2019-04-19:
In [ap6212-bluetooth](https://github.com/armbian/build/blob/2645877aa5d5c58ac792d74f1bdc751ede55fa8d/packages/extras/tools/ap6212-bluetooth#L59), rfkill is not used correctly.

```
        rfkill unblock all
        echo "0" > /sys/class/rfkill/rfkill0/state
        echo "1" > /sys/class/rfkill/rfkill0/state
```

for Nanopi Neo Air and Orangepi Zero Plus2 H5, rfkill0 is Wi-Fi in the default settings of armbian
```
pi@nanopiair:~$ rfkill list
0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
```

For Nanopi Neo Air,  I tried to add bluetooth reset to rfkill using a dts file, but failed. The dts is:
```
/dts-v1/;
/plugin/;

/ {
    compatible = "allwinner,sun8i-h3";

    fragment@0 {
        target = <&pio>;
        __overlay__ {
            bt_pwr_pin: bt_pwr_pin@0 {
                pins = "PG13";
                function = "gpio_out";
            };
        };
    };

    fragment@1 {
        target-path = "/";
        __overlay__ {
            rfkill_bt {
                compatible = "rfkill-gpio";
                pinctrl-names = "default";
                pinctrl-0 = <&bt_pwr_pin>;
                reset-gpios = <&pio 6 13 1>;
                clocks = <&osc32k>;
                clock-frequency = <32768>;
                rfkill-name = "sunxi-bt";
                rfkill-type = "bluetooth";
            };
        };
    };
};
```
The command `rfkill list` doesn't have "sunxi-bt". Something should be missing in the dts file.
Any idea?

igorpecovnik, 2019-04-19:
Bluetooth is in a bad shape in general so yes, its possible that things are missing. For me it's low priority ... but I guess once this should be sorted out. At least for AP6212, which is most common.

xiongyihui, 2019-04-20:
It seems that we need [a patch for rfkill-gpio](https://github.com/friendlyarm/linux/commit/de8acffc1a023f92f12fa9c9a4ff9656e9be14a2#diff-f03340c270d98b07cd53c0768aa766c8) to add bluetooth reset pin to rfkill-gpio

igorpecovnik, 2019-04-22:
Some progress:
```| \ | | __ _ _ __   ___ |  _ \(_) |  _ \/ |
|  \| |/ _` | '_ \ / _ \| |_) | | | |_) | |
| |\  | (_| | | | | (_) |  __/| | |  _ <| |
|_| \_|\__,_|_| |_|\___/|_|   |_| |_| \_\_|
                                           

Welcome to ARMBIAN 5.82 user-built Ubuntu 18.04.2 LTS 4.19.36-sunxi   
System load:   0.00 0.00 0.00  	Up time:       31 min		
Memory usage:  8 % of 999MB  	IP:            
CPU temp:      33°C           	
Usage of /:    3% of 30G    	

Last login: Mon Apr 22 11:24:25 2019 from 10.0.10.42

root@nanopi:~# rfkill list
0: sunxi-bt: Bluetooth
	Soft blocked: no
	Hard blocked: no
1: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
```

konkhra, 2021-04-02:
In Armbian Focal (mainline based kernel 5.10.y) the problem still persists. Bluetooth not working. Maybe something broken up and the question is what and where?

igorpecovnik, 2021-05-11:
> is this patch supposed to be included in a new release?

If not in the release documents and if issue not closed [with a commit](https://docs.armbian.com/Process_Contribute/), unlikely:
https://docs.armbian.com/Release_Changelog/

walterjj, 2021-05-15:
Could we have a workaround in teh while to make BLE work on a NanoPi NEO Air on Armbian 21.05.1 Focal with Linux 5.10.34-sunxi ?  Thanyou!

msabatini-met, 2021-05-18:
Give a look here to see how I found a solution:
https://forum.armbian.com/topic/18100-ap6212a-bt-not-working-on-orange-pi-zero-plus2-h3/
edit the ap6212-bluetooth script if needed for your platform.

leggewie, 2025-08-24:
FWIW, my BPi M2+ has the AP6212 and here is the output

$ rfkill list
0: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
1: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no

closed 2025-08-24 by igorpecovnik