armbian/build #1354: cannot build linux-source-4.19.20-mvebu64 (missing net/wireguard/Kconfig)

magnusgulin, 2019-04-21:
Seems like package linux-source-4.19.20-mvebu64 is also affected by the bug reported in #1278

```
root@espressobin:/usr/src/linux-source-4.19.20-mvebu64# make oldconfig
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  YACC    scripts/kconfig/zconf.tab.c
  LEX     scripts/kconfig/zconf.lex.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf  --oldconfig Kconfig
net/Kconfig:89: can't open file "net/wireguard/Kconfig"
scripts/kconfig/Makefile:69: recipe for target 'oldconfig' failed
make[1]: *** [oldconfig] Error 1
Makefile:539: recipe for target 'oldconfig' failed
make: *** [oldconfig] Error 2

```

igorpecovnik, 2019-04-21:
We need to rebuild and updated all kernels ... and that needs some testings at first since solving one problem can lead in opening new ones.

igorpecovnik, 2019-05-05:
Fixed with https://github.com/armbian/upload/commit/6c6897683f61201dc4f33a172288aabdb8581424