armbian/build #1358: U-Boot on dev branch, the ENV default to MMC

aprayoga, 2019-04-30:
https://github.com/armbian/build/blob/3c372b92a8bcd8c3ef0ee6e89a1d9f83dec0ff34/lib/compilation.sh#L157

```
		if [[ $BOOTBRANCH =~ ^tag:v201[8-9](.*) ]]; then
			[[ -f .config ]] && sed -i 's/^.*CONFIG_ENV_IS_IN_FAT.*/# CONFIG_ENV_IS_IN_FAT is not set/g' .config
			[[ -f .config ]] && sed -i 's/^.*CONFIG_ENV_IS_IN_EXT4.*/CONFIG_ENV_IS_IN_EXT4=y/g' .config
			[[ -f .config ]] && sed -i 's/^.*CONFIG_ENV_IS_IN_MMC.*/# CONFIG_ENV_IS_IN_MMC is not set/g' .config
			[[ -f .config ]] && sed -i 's/^.*CONFIG_ENV_IS_NOWHERE.*/# CONFIG_ENV_IS_NOWHERE is not set/g' .config | echo "# CONFIG_ENV_IS_NOWHERE is not set" >> .config
			[[ -f .config ]] && echo 'CONFIG_ENV_EXT4_INTERFACE="mmc"' >> .config
			[[ -f .config ]] && echo 'CONFIG_ENV_EXT4_DEVICE_AND_PART="0:auto"' >> .config
			[[ -f .config ]] && echo 'CONFIG_ENV_EXT4_FILE="/boot/boot.env"' >> .config
		fi
```

this block of code would overwrite environment setting into always as file /boot/boot.env regardless what the target is.

I believe if we built SPI flash image, we would expect the ENV also stored in SPI flash so no dependency with other device.

Would adding another condition (maybe if target != spi) acceptable?

igorpecovnik, 2019-05-02:
This block of code is temporally add until changes are not cleared out upstream and on all machines. I only add this for Allwinner since it was by the time the only one moved to 2018-19. Since we moved others here, this method need to be verified and checked with current defaults. Since nothing seems to break - if path is set wrong - I didn't make any changes. Perhaps this is not needed anymore?

aprayoga, 2019-05-02:
I'm working on Helios4 to test dev branch. I added spi target in UBOOT_TARGET_MAP and will put patch to modify helios4_defconfig to set the ENV in SPI flash (CONFIG_ENV_IS_IN_SPI_FLASH=y).

Expected .config before the compilation something like
```
# CONFIG_ENV_IS_IN_FAT is not set
# CONFIG_ENV_IS_IN_EXT4 is not set
# CONFIG_ENV_IS_IN_MMC is not set
# CONFIG_ENV_IS_NOWHERE is not set
CONFIG_ENV_IS_IN_SPI_FLASH=y
```

but after passing that block of code, it become
```
# CONFIG_ENV_IS_IN_FAT is not set
CONFIG_ENV_IS_IN_EXT4=y
# CONFIG_ENV_IS_IN_MMC is not set
# CONFIG_ENV_IS_NOWHERE is not set
CONFIG_ENV_IS_IN_SPI_FLASH=y
# CONFIG_ENV_IS_NOWHERE is not set
CONFIG_ENV_EXT4_INTERFACE="mmc"
CONFIG_ENV_EXT4_DEVICE_AND_PART="0:auto"
CONFIG_ENV_EXT4_FILE="/boot/boot.env"
```

I haven't tried to compile with this config but in my past experience it failed to compile when CONFIG_ENV_IS_IN_* defined in more than 1 device. Even if it compile, it would depend on mmc which is not the intention of using the SPI flash.
So I was wondering if you foresee any problem if I added condition to exclude spi target from executing that block since the changes would affect all board.

igorpecovnik, 2019-05-03:
> If the block removed, where the ENV stored in MMC? certain sector/offset between MBR and 1st partition?

On the file system. Due to space constrain this mess happen in 1st place. 

> So I was wondering if you foresee any problem if I added condition to exclude spi target from executing that block since the changes would affect all board.

Add mvebu* exception only and test. For others it has to be tested as well, especially for Allwinner, where we have this problem.

chwe17, 2019-06-07:
It might make sense to revamp all the defconfigs of allwinner in a u-boot patch. I remember this snippet fooled me once when working on mt7623 IIRC CONFIG_ENV_EXT4_DEVICE_AND_PART="0:auto" means eMMC for this SoC and https://github.com/armbian/build/blob/30aae1fc9c6f94c4447c6355bc918367f5741248/patch/u-boot/u-boot-mt7623/fix-dependencies-ugly-boot.patch#L88 was needed to 'duct tape' it (mt7623 was not even csc back then) but it took me nuts to find it. You don't expect 'patching outside of the patchfolders'. And it happens after patching so you can't revert it.

I would be in favor of a 'clean' fix.

aprayoga, 2019-07-01:
@igorpecovnik I looked at commit 648c52c, it still unclear for me, what is the intention.

My understanding,
From now on, ANY board that uses Mainline U-Boot 2018 or U-Boot 2019 would have the U-Boot Environment stored as a file (/boot/boot.env) on MMC/SD card. Except if the target is SPI image.

Could you confirm whether my understanding is correct?
I thought that block of code was meant for Allwinner only

Is there any other @armbian/board-maintainers would affected by this change?

igorpecovnik, 2019-07-01:
> Could you confirm whether my understanding is correct?

Correct, I only fix the problem for your case if you want to build spi target ... At least for Allwinner, this is AFAIK the only way, other boards has be checked if this method works. Alternatively it could be fixed with a patch.

aprayoga, 2019-08-01:
It seem commit 648c52c does not fix it. I'll investigate. here are the log on Helios4 Armbian 5.91
```
U-Boot 2019.04-armbian (Jul 31 2019 - 17:30:19 +0200)

SoC:   MV88F6828-B0 at 1600 MHz
DRAM:  2 GiB (800 MHz, 32-bit, ECC enabled)
MMC:   mv_sdh: 0
Loading Environment from EXT4... ** File not found /boot/boot.env **

** Unable to read "/boot/boot.env" from mmc0:1 **
Loading Environment from SPI Flash... SF: Detected w25q32 with page size 256 Bytes, erase size 4 KiB, total 4 MiB
*** Warning - bad CRC, using default environment

Model: Helios4
Board: Helios4
```

BTW, are those target name already standardized? I mean, for 
- SD card, the target name is *sdhc*
- eMMC, the target name is *mmc*
- SATA, the target name is *sata*
- SPI, the target name is *spi*

igorpecovnik, 2019-08-01:
It looks like it's just a typo target -> $target ... I'll fix it with next commit.

igorpecovnik, 2019-08-01:
> are those target name already standardized

Not completely sure, but let's try to be synchronised with u-boot terminology.

closed 2020-09-07 by aprayoga