armbian/build #1359: Cannot install firmware-ralink: firmware-misc-nonfree conlicts with armbian-firmware

jech, 2019-05-01:
```
$ sudo apt install firmware-ralink
...
Get:1 http://cdn-fastly.deb.debian.org/debian stretch/non-free arm64 firmware-misc-nonfree all 20161130-5 [2320 kB]
Get:2 http://cdn-fastly.deb.debian.org/debian stretch/non-free arm64 firmware-ralink all 20161130-5 [14.1 kB]
...
dpkg: error processing archive /var/cache/apt/archives/firmware-misc-nonfree_20161130-5_all.deb (--unpack):
 trying to overwrite '/lib/firmware/mt7601u.bin', which is also in package armbian-firmware 5.79.1
```

igorpecovnik, 2019-05-02:
It's fixed for the future versions. You don't need this firmware, but if you do ... uninstall armbian-firmware first.

closed 2019-05-02 by igorpecovnik

jech, 2019-05-03:
> You probably don't need this firmware

How else am I supposed to use an RT2573 USB stick?

> + Replaces: linux-firmware, firmware-brcm80211, firmware-ralink, firmware-samsung, firmware-realtek, armbian-firmware${REPLACE}

So now I can no longer use the RT2573 and keep armbian firmware?  Is that the intended result?
￼

igorpecovnik, 2019-05-03:
Install armbian-firmware-full or copy files manually.

jech, 2019-05-03:
Thanks for the info.  Where is that documented?

igorpecovnik, 2019-05-04:
Armbian-firmware contains files for on-board wifi chips and a few most common used ones. When you need something else, the problem becomes yours. I can only recommend you to use either a full package (armbian-firmware-full) or cherry pick on your own. Sometimes this could mean deinstalling armbian-firmware.

There is no manual for manual way since it depends from the particular firmware. Not exactly an Armbian problem, but generic ... which is far out of our power to help with.

jech, 2019-05-05:
> There is no manual for manual way since it depends from the particular
> firmware. Not exactly an Armbian problem, but generic ... which is far out of
> our power to help with.

I understand that.

What I think should be documented is the relationship between
armbian-firmware, armbian-firmware-full, and the upstream firmware
packages.  Roughly speaking, the explanation you gave above should be
copy-pasted into the documentation (together with instructions on how to
switch from armbian-firmware to the upstream firmware packages).