armbian/build #1360: 100% CPU load with armbian-truncate-logs

carlosnewmusic, 2019-05-05:
the script generates 178 processes armbian-truncate-logs, which in turn many children processes (89 armbian-ramlog write) which in turn generates other children processes (rsync -aXWv --delete --exclude armbian-ramlog.log - links / var / log / /var/log.hdd/ and tee -a /var/log.hdd/armbian-ramlog.log) of more and so I'm seeing this it does for every system log, which makes that in total among all the processes, subprocesses and threads of threads come to occupy 100% of CPU usage

igorpecovnik, 2019-05-07:
How can this be recreated?

carlosnewmusic, 2019-05-10:
> How can this be recreated?

filling all or enough logs to more than 75% and when the cleaning process starts it should happen

igorpecovnik, 2020-01-03:
> filling all or enough logs to more than 75% and when the cleaning process starts it should happen

Was doing this right now on latest image. It doesn't happen on latest self-made images.

closed 2020-01-03 by igorpecovnik