armbian/build #1367: [RK3288] Confusing "Built-in Audio" Pulseaudio devices

paolosabatino, 2019-05-11:
Hello,

I'm using my RK3288 box as a general all-purpose desktop replacement device.
As desktop replacement, I find the "Built-In Audio" pulseaudio labels for both SPDIF and HDMI audio devices annoying and confusing.

I would like to propose an asound.conf for RK3288 devices as given below that adds some ALSA configuration to expose HDMI and SPDIF devices with proper tags.
This allows PulseAudio to show the labels correctly instead of the standard and confusing "Built-in Audio" for both devices and possibly enable some kind of passthrough (not tested).

A previous unanswered forum post with a unpolished version of this patch, with some more explanation and a screenshot of the result, is here: https://forum.armbian.com/topic/10183-pulseaudio-and-built-in-audio-issue-involves-alsa/

Audio devices got the correct ALSA labels only if their name is SPDIF/DW-I2S-HDMI in devicetree. Tinkerboards are using "rockchip,tinker-codec" for HDMI device so this patch won't work on them. 

I was wondering if the tinkerboard label is proper because device trees for other devices does not use the "vendor,device" nomenclature for simple-sound-card,name property. Changing it to DW-I2S-HDMI can make Tinkerboards benefit from this patch. 

I can arrange a merge request for both the asound.conf patch and tinkerboard audio card name fix, but I'm open to any suggestion.

@Tonymac32 

```
pcm.OnBoard_D0 {

        type hw
        card OnBoard
        device 0

}

 

pcm.OnBoard_D1 {

        type hw
        card OnBoard
        device 1

}

 

pcm.OnBoard_D2 {

        type hw
        card OnBoard
        device 2

}

# Configuration to expose the SPDIF device of RK3288 to ALSA.
# The device tree must describe a simple-audio-card device named "SPDIF"
# for this to work properly
# This percolates to PulseAudio, which will eventually show the proper
# labels to the audio device

<confdir:pcm/iec958.conf>

cards.SPDIF.pcm.iec958.0 {
        @args [ CARD AES0 AES1 AES2 AES3 ]
        @args.CARD {
                type string
        }
        @args.AES0 {
                type integer
        }
        @args.AES1 {
                type integer
        }
        @args.AES2 {
                type integer
        }
        @args.AES3 {
                type integer
        }
	type hw
	card $CARD

}

# Configuration to expose the HDMI device of RK3288 to ALSA.
# The device tree must describe a simple-audio-card device named "DW-I2S-HDMI"
# for this to work properly
# This percolates to PulseAudio, which will eventually show the proper
# labels to the audio device

<confdir:pcm/hdmi.conf>

cards.DW-I2S-HDMI.pcm.hdmi.0 {
        @args [ CARD DEVICE CTLINDEX AES0 AES1 AES2 AES3 ]
        @args.CARD {
                type string
        }
        @args.DEVICE {
                type integer
        }
        @args.CTLINDEX {
                type integer
        }
        @args.AES0 {
                type integer
        }
        @args.AES1 {
                type integer
        }
        @args.AES2 {
                type integer
        }
        @args.AES3 {
                type integer
        }
        type hw
	card $CARD 
}
```

Tonymac32, 2019-05-11:
The tinker board settings are from Asus and match their use on Tinker OS.  It's also important to remember the Tinker's audio is provided via a USB DAC built onto the board, so I think only the gpio I2S and HDMI I2S are via the SoC.  there is an associated USB tweak to name the DAC to accommodate that config file.  Combine these things and that's why it doesn't seem the Tinker settings make sense in context with RK3288 hardware.

I can check your patch, there would probably have to be some logic added to the build scripts in the event it messes with the Tinker.

paolosabatino, 2019-05-11:
Thanks for explanation. 

The asound.conf above is a carbon-copy for the first part (the one describing pcm.OnBoard_Dx devices) so it should not harm the analog outputs provided by the USB DAC.

It just adds the two sections for the digital SPDIF and HDMI outputs provided directly by the SoC.
About _rockchip,tinker-codec_, I was referring to the _sound_ section in tinkerboard.dtsi (see https://github.com/torvalds/linux/blob/1fb3b526df3bd7647e7854915ae6b22299408baf/arch/arm/boot/dts/rk3288-tinker.dtsi#L64): there seems to be a confusion with _simple-audio-card,name_ property  because that section clearly relates to the SoC HDMI digital output, but looking at the name it seems to relate to the Tinkerboard analog codec.

Tonymac32, 2019-05-11:
Alright, I can look at it/test it once I get to my PC later, I was reviewing it on my phone, so not ideal.  Most of this was pulled right from the 4.4 kernel or mainline without adjustment, and as usual I would expect the Rockchip reference board stuff to be mixed in with the actual design, probably with some improper mixing.  Thanks for looking into this, I'll give it some tests

paolosabatino, 2019-06-07:
@Tonymac32 Did you have the chance to try the snippet?

Tonymac32, 2019-06-09:
Trying now, it's been an insane few weeks

Tonymac32, 2019-06-09:
Done.  Looks much nicer, I agree.

closed 2019-06-09 by Tonymac32