armbian/build #1369: Cannot build DKMS modules for sunxi-next

mkudlacek, 2019-05-14:
I have OrangePi R1 running Debian Stretch:

`Linux orangepi 4.19.38-sunxi #5.86 SMP Sun May 12 17:56:25 CEST 2019 armv7l GNU/Linux`

I installed headers of the same version:

```
ii  linux-headers-next-sunxi            5.86                           armhf        Linux kernel headers for 4.19.38-sunxi on armhf
ii  linux-image-next-sunxi              5.86                           armhf        Linux kernel, version 4.19.38-sunxi
```
When I try building a module via DKMS, it ends up with an error:

```
root@orangepi:~# dpkg-reconfigure wireguard-dkms

------------------------------
Deleting module version: 0.0.20190406
completely from the DKMS tree.
------------------------------
Done.
Loading new wireguard-0.0.20190406 DKMS files...
Building for 4.19.38-sunxi
Building initial module for 4.19.38-sunxi
Error! Bad return status for module build on kernel: 4.19.38-sunxi (armv7l)
Consult /var/lib/dkms/wireguard/0.0.20190406/build/make.log for more information.
```

The mentioned log file contains:

```
DKMS make.log for wireguard-0.0.20190406 for kernel 4.19.38-sunxi (armv7l)
Tue May 14 10:36:54 UTC 2019
make: Entering directory '/usr/src/linux-headers-4.19.38-sunxi'
Makefile:619: arch//Makefile: No such file or directory
make: *** No rule to make target 'arch//Makefile'.  Stop.
make: Leaving directory '/usr/src/linux-headers-4.19.38-sunxi'
```

It looks like the $SRCARCH is empty (line 619 of the mentioned Makefile):

```
# The arch Makefile can set ARCH_{CPP,A,C}FLAGS to override the default
# values of the respective KBUILD_* variables
ARCH_CPPFLAGS :=
ARCH_AFLAGS :=
ARCH_CFLAGS :=
include arch/$(SRCARCH)/Makefile
```

I tried to force the architecture by following suggestions on this page: https://github.com/diederikdehaas/rtl8812AU/issues/103

Namely:

```
ARCH=arm dpkg-reconfigure wireguard-dkms
```

Or adding the symlink:

```
root@orangepi:/usr/src/linux-headers-4.19.38-sunxi/arch# ls -l
total 124
drwxr-xr-x   9 root root  4096 May 13 12:42 alpha
drwxr-xr-x  13 root root  4096 May 13 12:42 arc
drwxr-xr-x 100 root root  4096 May 13 12:42 arm
drwxr-xr-x  11 root root  4096 May 13 12:42 arm64
lrwxrwxrwx   1 root root     3 May  8 18:46 armv7l -> arm
```

Or even changing the line 619 in the /root@orangepi:/usr/src/linux-headers-4.19.38-sunxi/Makefile to:

```
include arch/arm/Makefile
```

But none of it worked. I tried different DKMS module (virtualbox-dkms), but obviously, the problem is in the linux-headers-next-sunxi package.

What can I do to make it work?

Thanks,
Martin

igorpecovnik, 2019-05-14:
Wireguard is part of all Armbian kernels. That's why you can't build. Just run:

`modprobe wireguard`

mkudlacek, 2019-05-14:
Hi @igorpecovnik,
 thanks for the reply. It fixed my problem - I am able to run the wireguard now:) However, I don't think the problem is related to the wireguard module already being part of the kernel.  The same error happens when I try to build the virtualbox dkms module:

```
root@orangepi:/# dpkg-reconfigure virtualbox-dkms

------------------------------
Deleting module version: 5.2.24
completely from the DKMS tree.
------------------------------
Done.
Loading new virtualbox-5.2.24 DKMS files...
Building for 4.19.38-sunxi
Building initial module for 4.19.38-sunxi
Error! Bad return status for module build on kernel: 4.19.38-sunxi (armv7l)
Consult /var/lib/dkms/virtualbox/5.2.24/build/make.log for more information.
```

Content of the mentioned log:
```
DKMS make.log for virtualbox-5.2.24 for kernel 4.19.38-sunxi (armv7l)
Tue May 14 12:32:51 UTC 2019
make: Entering directory '/usr/src/linux-headers-4.19.38-sunxi'
Makefile:619: arch//Makefile: No such file or directory
make: *** No rule to make target 'arch//Makefile'.  Stop.
make: Leaving directory '/usr/src/linux-headers-4.19.38-sunxi'
```

But the module is not available:

```
root@orangepi:/# modprobe vboxdrv
modprobe: FATAL: Module vboxdrv not found in directory /lib/modules/4.19.38-sunxi
```

igorpecovnik, 2019-05-14:
Nice, and how did you solve it?

mkudlacek, 2019-05-14:
The wireguard by modprobing the module: 

```modprobe wireguard```

The virtualbox I did not. I don't actually need it, I just wanted to demonstrate, that it is probably some another issue than the one with wireguard.

igorpecovnik, 2019-05-14:
Aha, ok ... IIRC DKMS was broken upstream for some time and since its impossible to fix everything it will remain as is until someone would desperately need this function :) Since DMKS is apparently broken I'll leave this open.

devinrsmith, 2019-05-23:
Ditto on ROCKPro64, getting similar error trying to install wireguard-dkms:

```
DKMS make.log for wireguard-0.0.20190406 for kernel 4.4.178-rockchip64 (aarch64)
Thu May 23 16:46:08 UTC 2019
make: Entering directory '/usr/src/linux-headers-4.4.178-rockchip64'
Makefile:666: arch//Makefile: No such file or directory
make: *** No rule to make target 'arch//Makefile'.  Stop.
make: Leaving directory '/usr/src/linux-headers-4.4.178-rockchip64'
```

pianoquintet, 2019-07-11:
Same on an Odroid HC1 with armbian stretch.  "modprobe wireguard" returns no errors but I am left without all the wireguard commands (wg-quick, wg, wg-show, etc.)

igorpecovnik, 2019-07-11:
> Same on an Odroid HC1 with armbian stretch. "modprobe wireguard" returns no errors but I am left without all the wireguard commands (wg-quick, wg, wg-show, etc.)

If you care to add and maintain (usually nothing) the build of wg packages https://github.com/armbian/build/tree/master/packages/extras-buildpkgs 

Or those tools already exists in some repository?

zhangn1985, 2019-08-13:
Hi, @igorpecovnik please due to I rebase sunxi-next to upstream LTS, please set `WIREGUARD=yes` to enable wireguard for sunxi-next release, if wireguard is required feature.

pianoquintet, 2019-08-13:
Where do I place this setting? Thanks.

igorpecovnik, 2019-08-13:
> Where do I place this setting? Thanks.

Just build image, now WG is default. For the DKMS way I can't help without deep investigation ...

zhfreal, 2019-09-09:
Hey,
I fixed this issue by rebuild headers in armbian device. The orginal headers cross builded would never working. I copy the whole kernel source to armbian, then "make prepare LOCALVERSION='-xxxx'",  then "make scripts LOCALVERSION='-xxxx'". The LOCALVERSION is the postfix str of kernel version. Ex, kernel '5.2.1-aml-g12', the LOCALVERSION is '-aml-g12'. The LOCALVERSION is very important. And link source dirctory to /lib/modules/$(uname -r)/build.  After that, dkms is working perfectly. But you need 1 more gigabytes free space in your armbian device. And I have no idea to how to reduce it.

mmriech, 2019-12-28:
Pull request #1697 is supposed to fix this issue. Please retest as soon as the PR is merged.

count-doku, 2019-12-28:
I just merged it. Please rebuild and test.

igorpecovnik, 2020-01-07:
Tested, works as expected.

closed 2020-01-07 by igorpecovnik