armbian/build #1375: Cant find tritium H2+/H3 in build targets

Ntemis, 2019-05-16:
Hi,
Am currently using my Tritium H2+ as a DVB-T/C/S USB server  with minisatip https://github.com/catalinii/minisatip.
I have recently purchased 4x Microsoft Xbox One Digital TV Tuner Digital TV tuner DVB-C, DVB-T2 USB and i would like to use them on 5.x kernel just to get every last bug squashed :)
I cloned the repo and tried to build tritium h3 as target but i cant find it anywhere, please advice.
Regards

Tonymac32, 2019-05-16:
I just checked the folder and the config is gone.  WTF?  I'll figure out when and how it happened and restore it, but it won't be until later.

igorpecovnik, 2019-05-16:
My mistake. Fixing it.

closed 2019-05-16 by igorpecovnik

ThomasKaiser, 2019-05-16:
> Fixing it.

The configs were either CSC or WIP before: https://github.com/armbian/build/commit/25889024ddfe8282bf8ffa41f5ea200a6d147a12#diff-c80a6818f132d1b13b2b125aa2af4aed

Tonymac32, 2019-05-16:
> The configs were either CSC or WIP before: 2588902
> #diff-c80a6818f132d1b13b2b125aa2af4aed

Yes, they were, but since they are nothing more than breakout boards for
the SoC there was some discussion about marking them supported (images
around for some time for some reason despite WIP/CSC, without complaints)
so maybe someone renamed them and didn't add the renamed file when they
committed.

igorpecovnik, 2019-05-16:
> didn't add the renamed file when they committed.

Yes, this was the problem. If Pinebook is .conf Teres can also be. It has even better mainline support atm ... Banana M64 also works the same good or bad as other A64 boards. If this is not ok, revert. I will not fight for this.

Ntemis, 2019-05-16:
hmmm Show a kernel configuration menu before compilation doesn't work sorry guys.
Still pulls linux-mainline orange-pi-4.19

igorpecovnik, 2019-05-16:
Everything is working properly ... It seems your have either wrong/own user configuration or some corruption in the script. Check or just start from a clean fork.