armbian/build #1388: missing lima driver (sunxi64 dev)

digitallyelite, 2019-05-30:
When booting linux-image-dev-sunxi64 (both 5.1.3 and 5.1.5) - it appears that the lima module isn't enabled.  I'm not sure if this is an oversight, or on purpose - Looking in the linux-source-5.1.5-sunxi64 tarball (and the patches in the repo) it appears to be patched in to the kernel, but was removed by commit b50b7e685ef96dd7d146f86ef5a149e4790e5d28

igorpecovnik, 2019-05-30:
> but was removed by 

Sine adjustments represent a lot of work and this is not a vital part. If you don't have a picture, this is 100% not due missing lima driver.

If you need Lima driver for some reason in development kernel you are welcome to adjust patches, but since it seems Lima is officially getting mainline with 5.2.y this makes little sense. We will certainly not blow time for this.

I boot Teres (which is very similar to Pinebook) with 5.1.y yesterday and it worked fine (screen).

BTW. This is build script issue tracker. DEVelopment discussuins shell go [here](https://forum.armbian.com/forum/20-development/) - no end-user support. If it works, works, if not, you are on your own to fix/find out.

closed 2019-05-30 by igorpecovnik

digitallyelite, 2019-05-31:
Hi Igor...
I'm asking for the LIMA option to be turned on in the kernel config, that's the request.  I'm not asking for user support.  The LIMA patches are already applied, but the kernel config option was disabled in that commit, not the patches were removed.

igorpecovnik, 2019-05-31:
User support. Communicating with random people inside and around that wants something represents a bulk of time we waste on the project. Request is denied - you are on your own - [send a prepared and tested PR](https://docs.armbian.com/Process_Contribute/). Contributions are welcome but they have to be meaningful to be accepted! This one is beyond common interest and you didn't prepare anything to save time.