armbian/build #1396: sunxi-dev kernel does not compile

EvilOlaf, 2019-06-02:
`dvbskys960-fix.patch`  failing to apply.

debug compilation log:

```
drivers/media/usb/dvb-usb-v2/dvbsky.c:842:22: error: ‘dvbsky_exit’ undeclared here (not in a function); did you mean ‘dvbsky_init’?
  .exit             = dvbsky_exit,
                      ^~~~~~~~~~~
                      dvbsky_init
make[4]: *** [drivers/media/usb/dvb-usb-v2/dvbsky.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [drivers/media/usb/dvb-usb-v2] Error 2
```

I am not perfectly sure but I think to remember that 5.1.5 compiled successfully.
I dug a bit further and noticed there were changes from 5.1.5 to 5.1.6 regarding dvbsky.
https://github.com/megous/linux/commit/0154f03e43b58dc3957dfca45587e227c24f951f

igorpecovnik, 2019-06-02:
Watch this https://github.com/armbian/build/commit/f7edc91c8fad7bda2ec591de9ab237da0e559688

EvilOlaf, 2019-06-02:
I noticed that one, but did not give further attention as sunxi-next compiling fine for me.

igorpecovnik, 2019-06-02:
dvbsky patches landed mainline and they broke some of ours. The problem is the same as it was in sunxi-next branch. I will fix it later on but if you got the time ... its just removing a patch and one line in second patch.

EvilOlaf, 2019-06-02:
I can try but will probably fail. I am not very deep in coding at all, but I try to be as helpful as possible to track down issues with my basic knowledge.

igorpecovnik, 2019-06-02:
https://github.com/armbian/build/commit/61f795384266124746e2ad71307832f99b3f546f

closed 2019-06-02 by igorpecovnik