armbian/build #1397: Enable CONFIG_NETFILTER_XT_SET for odroidn2

digaxfr, 2019-06-02:
This is a a feature request (or maybe even a bug/mis-configuration?) to enable `CONFIG_NETFILTER_XT_SET=m` for the odroidn2 board.

I have compiled this myself using `linux-source-4.9.179-default-odroidn2` and it works as expected. This is very useful for use cases such as Kubernetes using Calico CNI. Calico requires `XT_SET`.

On a fresh clone of the repo, we can actually see this is set in every kernel config except for  `linux-odroidn2-default.config` and `linux-sun5i-default.config`.

```
[dchin@failxps kernel]$ for i in $(ls); do echo -n $i-; grep CONFIG_NETFILTER_XT_SET $i; echo; done
linux-cubox-default.config-CONFIG_NETFILTER_XT_SET=m
linux-cubox-next.config-CONFIG_NETFILTER_XT_SET=m       
linux-meson64-default.config-CONFIG_NETFILTER_XT_SET=m
linux-meson64-dev.config-CONFIG_NETFILTER_XT_SET=m   
linux-meson64-next.config-CONFIG_NETFILTER_XT_SET=m  
linux-meson-default.config-CONFIG_NETFILTER_XT_SET=m 
linux-mt7623-default.config-CONFIG_NETFILTER_XT_SET=m    
linux-mvebu64-default.config-CONFIG_NETFILTER_XT_SET=m
linux-mvebu64-dev.config-CONFIG_NETFILTER_XT_SET=m     
linux-mvebu64-next.config-CONFIG_NETFILTER_XT_SET=m
linux-mvebu-default.config-CONFIG_NETFILTER_XT_SET=m
linux-mvebu-dev.config-CONFIG_NETFILTER_XT_SET=m   
linux-mvebu-next.config-CONFIG_NETFILTER_XT_SET=m   
linux-odroidc1-default.config-CONFIG_NETFILTER_XT_SET=m                                                    
linux-odroidc1-next.config-CONFIG_NETFILTER_XT_SET=m
linux-odroidn2-default.config-# CONFIG_NETFILTER_XT_SET is not set
linux-odroidn2-next.config-CONFIG_NETFILTER_XT_SET=m
linux-odroidxu4-default.config-CONFIG_NETFILTER_XT_SET=m
linux-odroidxu4-dev.config-CONFIG_NETFILTER_XT_SET=m
linux-odroidxu4-next.config-CONFIG_NETFILTER_XT_SET=m
linux-pine64-default.config-CONFIG_NETFILTER_XT_SET=m
linux-rk3399-default.config-CONFIG_NETFILTER_XT_SET=m
linux-rockchip64-default.config-CONFIG_NETFILTER_XT_SET=m
linux-rockchip64-dev.config-CONFIG_NETFILTER_XT_SET=m
linux-rockchip-default.config-CONFIG_NETFILTER_XT_SET=m
linux-rockchip-dev.config-CONFIG_NETFILTER_XT_SET=m
linux-rockchip-next.config-CONFIG_NETFILTER_XT_SET=m
linux-s5p6818-next.config-CONFIG_NETFILTER_XT_SET=m
linux-sun4i-default.config-CONFIG_NETFILTER_XT_SET=m
linux-sun5i-default.config-
linux-sun7i-default.config-CONFIG_NETFILTER_XT_SET=m
linux-sun8i-default.config-CONFIG_NETFILTER_XT_SET=m
linux-sun8i-dev.config-CONFIG_NETFILTER_XT_SET=m
linux-sunxi64-dev.config-CONFIG_NETFILTER_XT_SET=m
linux-sunxi64-next.config-CONFIG_NETFILTER_XT_SET=m
linux-sunxi-dev.config-CONFIG_NETFILTER_XT_SET=m
linux-sunxi-next.config-CONFIG_NETFILTER_XT_SET=m
linux-udoo-next.config-CONFIG_NETFILTER_XT_SET=m
```

This looks fairly straight forward, and I would be happy to open the pull request to enable this on odroidn2.

igorpecovnik, 2019-06-02:
> This looks fairly straight forward, and I would be happy to open the pull request to enable this on odroidn2.

Do it.

digaxfr, 2019-06-02:
Merged https://github.com/armbian/build/pull/1398

closed 2019-06-02 by digaxfr