armbian/build #1406: How to install qemu for arm7v32 on aarch64 image builder?

NicolasDorier, 2019-06-07:
My apologize if I don't find the right terminology to explain my goal:

I am building customized armbian image for rock64, which is aarch64. Now, some software which will run on it are arm32v7. 

Now there is no problem into running such software on the actual rock64, because the kernel is built with `CONFIG_COMPAT` to `true`.

However, I need to run such armv7 program in `customize-script.sh`.

It seems that the way the armbian build work is by installing qemu aarch64 outside of chroot in https://github.com/armbian/build/blob/30aae1fc9c6f94c4447c6355bc918367f5741248/lib/chroot-buildpackages.sh .

Is there an easy way to add arm7v32 ?

NicolasDorier, 2019-06-07:
I was thinking to install qemu static inside the `customize-script` is it the right approach?

lanefu, 2019-06-25:
@NicolasDorier actually I think you can add the 32bit userspace libraries and that will be probably more ideal.. otherwise qemu is running inside of qemu.

something like
```
dpkg --add-architecture armhf
apt-get install libc6:armhf libstdc++6:armhf
```

lanefu, 2019-06-26:
closing, can reopen if needed

closed 2019-06-26 by lanefu

NicolasDorier, 2019-06-26:
@lanefu the problem was running the executable, not downloading the packages.

My solution was to the extension point `customize-image-host.sh`:

```bash
#!/bin/bash

display_alert "Add BTCPay Box customization" "qemu-arm-static" "info"
cp /usr/bin/qemu-arm-static "${SDCARD}"/usr/bin/
```