armbian/build #1426: Move mvebu DEFAULT, NEXT and Dev branch to next kernel (LTS) and U-boot

g-provost, 2019-06-26:
- Move Armbian Default to 4.14 Mainline + Uboot 2018
- Move Armbian Next to 4.19 Mainline + Uboot 2019
- Move Armbian Dev to 5.x Mainline + Uboot 2019

Board impacted :
* Helios4
* Clearfog Pro/Base

aprayoga, 2019-06-26:
@g-provost Does **Default** branch for Clearfog also need to be changed to 4.14 and U-Boot 2018?
I prefer to keep the default as it is now until someone from SolidRun change it.

g-provost, 2019-06-26:
@aprayoga We are already sharing the same kernel so there is no reason we can't the move for all mvebu. As for the U-Boot we can keep it segregated as it is already.

igorpecovnik, 2019-06-26:
Current default is 4.4.y which is IMO obsolete for some time and shell be removed, yes. Current (mainline + patches) next becomes default (we don't need legacy 4.14.y or?) ... while legacy/next u-boot is actually 2013 https://github.com/SolidRun/u-boot-armada38x I did few tests with a modern 2018/2019 but was long time ago. This means latest u-boot with both next and dev.

g-provost, 2019-06-26:
Yes we will do as the description says
- Move Armbian Default to 4.14 Mainline + Uboot 2018
- Move Armbian Next to 4.19 Mainline + Uboot 2019
- Move Armbian Dev to 5.x Mainline + Uboot 2019

It's just that for clearfog we need someone to do the work for u-boot 2018 and 2019.

g-provost, 2019-06-26:
Actually for 2018 it's already ready : https://github.com/SolidRun/u-boot/tree/v2018.01-solidrun-a38x

g-provost, 2019-06-26:
@aprayoga Is your worry related to armbian boot script support by u-boot from solidrun ?

igorpecovnik, 2019-06-26:
> Actually for 2018 it's already ready : https://github.com/SolidRun/u-boot/tree/v2018.01-solidrun-a38x

Aha, didn't notice. Zador made Clearfog implementation back then to have support for SD, eMMC, UART, M2 boot ... it would be nice to keep that. That's I am worrying about.

g-provost, 2019-06-27:
We will only upgrade U-boot if we manage to keep all the booting option supported so far. Same question apply to Helios4 with SPI boot since @aprayoga is already facing challenge with U-boot 2019. We shall see by end of the week.

aprayoga, 2019-06-28:
SPI support on U-boot 2019.04 broken since this commit https://github.com/u-boot/u-boot/commit/30c4383da3fa2652e551b0610e71360394aa51e1

igorpecovnik, 2019-06-28:
> SPI support on U-boot 2019.04 broken since this commit u-boot/u-boot@30c4383 (v2019.04-rc1)

Good to know. We have to keep at least one (2018?) capable of SPI boot ... perhaps this will be fixed in 2019.07?

igorpecovnik, 2019-06-30:
Let's stay for now on stock Solidrun U-boot 2018.11. It seems alright ... Clearfog PRO was tested briefly, now fine-tuning, testing, fixing and implementing Helios4.

aprayoga, 2019-07-01:
@igorpecovnik , are you going change helios4? please hold it. I'm working on helios4. I will adapt to clearfog configuration.

I have fixed the SPI issue on 2019.04 U-Boot.

igorpecovnik, 2019-07-01:
No, I wait for your changes.

> I have fixed the SPI issue on 2019.04 U-Boot.

Cool. Then attach NEXT and DEV to it and I will test on Clearfog. I hope we will be able to make a change this week, next I don't have access to hardware.

aprayoga, 2019-07-01:
@g-provost 

> Move Armbian Default to 4.14 Mainline + Uboot 2018

**Linux** Kernel switched to **4.14** on commit https://github.com/armbian/build/commit/4f685a6fe0c05af4041b89091f818d5abdfa49a4 by @igorpecovnik 

**U-Boot** switched to **v2018.01** on commit https://github.com/armbian/build/commit/4f685a6fe0c05af4041b89091f818d5abdfa49a4 by @igorpecovnik for Clearfog (source pointed to SolidRun's repo)
**U-Boot** switched to **v2018.11** on commit https://github.com/armbian/build/commit/e7c2942b1da7c714240bb82af524557eda646cd6 for Helios4 (source pointed to Helios4 repo)

g-provost, 2019-07-01:
@aprayoga ok so this is for DEFAULT branch.You are doing NEXT branch now ? If you can NEXT branch should use 2019.04 for both clearfog and Helios4.

aprayoga, 2019-07-01:
@igorpecovnik , I'm working on U-Boot 2019.04 under dev branch.
I put the patch to re-enable SPI support on this commit https://github.com/armbian/build/commit/74b95ba16f16704f7b071114aad84a0c76d7d70f
I'm not sure if I need to put it in NEXT as it based on SolidRun's v2018.01 which doesn't seem to have this problem

g-provost, 2019-07-01:
@igorpecovnik Please test the @aprayoga U-boot 2019.05 patch on Clearfog. This way we can decide if we move NEXT branch to 2019.04 for both Helios4 and Clearfog.

Also @aprayoga rise a good point, upgrade path (through apt upgrade) should be tested.

g-provost, 2019-07-01:
BTW another question, shouldn't we leave Debian 9 Stretch as CLI_TARGET for DEFAULT ? Debian 9 Stretch is LTS until 2022.

igorpecovnik, 2019-07-01:
I am out office atm and I am not sure my Clearfog has a SPI flash on it. Can't test that feature. Debian Buster is under testing and things just work: https://github.com/armbian/config#software-testings ... I guess those are minor problems from our perspective. I would vote to move on to Buster.

g-provost, 2019-07-03:
Ok then let's go for Buster for all branches then. We should take in consideration dist-upgrade path for user running Stretch.

@igorpecovnik So does your Clearfog has SPI ?

igorpecovnik, 2019-07-03:
> dist-upgrade

... is users own risk until today and if will remain this way. Feel free to test for your board and note that. So far, it worked and I don't expect troubles.

No SPI chip.

aprayoga, 2019-07-05:
initial testing buster-next on helios4.
- lm75 compiled as kernel module but not loaded during runtime.
- helios4 udev rules broken for cpu temp so the fancontrol failed to start. the device name changed from **armada_thermal** into **f10e4078.thermal**
- warning/info from systemd regarding fancontrol:
```
systemd[1]: /lib/systemd/system/fancontrol.service:9: PIDFile= references path below legacy directory /var/run/, updating /var/run/fancontrol.pid \xe2\x86\x92 /run/fancontrol.pid; please update the unit file accordingly.
```
- WoL working
- ATA leds working
- USB & Heartbeat leds working

aprayoga, 2019-07-08:
Testing upgrade/update scenario from current NEXT branch **Armbian_5.75_Helios4_Debian_stretch_next_4.14.98**.
Local network test repo created and added to the system and then run
`sudo apt upgrade`

Kernel updated to **4.19.56-mvebu** and following issue also occurred
- lm75 compiled as kernel module but not loaded during runtime.
- helios4 udev rules broken for cpu temp so the fancontrol failed to start. the device name changed from **armada_thermal** into **f10e4078.thermal**

The fixes mentioned on previous comment does not applied to the system.

`sudo apt dist-upgrade` did not upgrade the system.
```
root@helios4:~# apt dist-upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@helios4:~#
```

igorpecovnik, 2019-07-08:
> did not upgrade the system.

You mean kernel package did upgrade while board bsp didn't?

aprayoga, 2019-07-08:
on previous comment, i did
`apt dist-upgrade`
after
`apt upgrade`
and nothing happened.

---
Today i redo the test, from fresh install **Armbian_5.75_Helios4_Debian_stretch_next_4.14.98** then do
`apt dist-upgrade`. _(After adding local test repo, created using aptly)_

The packages updated but still on Stretch.

```
 _   _      _ _           _  _
| | | | ___| (_) ___  ___| || |
| |_| |/ _ \ | |/ _ \/ __| || |_
|  _  |  __/ | | (_) \__ \__   _|
|_| |_|\___|_|_|\___/|___/  |_|

Welcome to Debian Stretch with Armbian Linux 4.19.56-mvebu
System load:   0.38 0.10 0.03   Up time:       0 min
Memory usage:  2 % of 2016MB    IP:            192.168.0.113
CPU temp:      42°C
Usage of /:    4% of 30G

New to Armbian? Check the documentation first: https://docs.armbian.com
Last login: Mon Jul  8 06:24:59 2019 from 192.168.0.181

root@helios4:~#
```

Note: I did not set/modified the repo source to point to buster repo as instructed on
https://www.debian.org/releases/stable/armhf/release-notes/ch-upgrading.en.html#upgrade-process

g-provost, 2019-07-08:
@aprayoga Unfortunately there is no trivial way like in Ubuntu to upgrade Debian version in one command. We will have to give instruction to user running Stretch on how they could upgrade to Buster  (at their own risk). As you suggested we could point them to this : https://www.debian.org/releases/stable/armhf/release-notes/ch-upgrading.en.html#upgrade-process

igorpecovnik, 2019-07-08:
Debian repostory is tagged to release, not to status (stable)
https://github.com/armbian/build/blob/master/lib/general.sh#L138
That's why dist-upgrade doesn't work. Shell we change this?

igorpecovnik, 2019-07-08:
Perhaps by adding:

if $RELEASE = bionic || buster then release="stable" else release=$RELEASE

aprayoga, 2019-07-08:
I'll change the repo to stable and test it

g-provost, 2019-07-08:
@igorpecovnik I wouldn't recommend to use status name, it can be confusing. Plus anyway it's too late to do that now since existing install have their APT pointing to release name.

igorpecovnik, 2019-07-08:
> I wouldn't recommend to use status name, it can be confusing.

True. Leave as is then.

g-provost, 2019-07-09:
@igorpecovnik  Actually since Buster ship with LK 4.19, wouldn't it make more sense that for LK 4.14 (Armbian Default branch) we stick to Stretch ?  This way it give the choice to user to still use Stretch since some 3rd party app might not yet be fully supported on Buster. What do you think ?

igorpecovnik, 2019-07-09:
We never were attached with a kernel version upstream and we don't need now. Armbian kernel is not near the same as the one that is shipped with Buster. Well, in case of this particular hardware is not that changed, but still - configuration is ours, there are small changes and there are additional drivers which are not present in Buster stock kernel. We ship user space with best kernel and different user space is here only to give users an option to stay behind for just any reason they have and loose nothing hardware/low level security wise.

If one for some reason wants older kernel he can change that freely: armbian-config -> alternative kernels

g-provost, 2019-07-09:
Ok understood. But then from user perspective and looking at mvebu situation, if both Default and Next image are based on Buster, what would be the reason for a user to choose the Default image since both 4.14 and 4.19 kernel provide full support of the hardware ? 

I think for Helios4 we will provide Default / Stretch along Next / Buster. The reason is that some 3rd party app (provided through armbian-config) don't seem to install / work yet on Buster.

Also I'm a bit lost on the purpose of CLI_TARGET=  variable in board config. For example in helios4.conf we have
`CLI_TARGET="buster,bionic:next"`
What does CLI_TARGET= does exactly ? Because I can still choose Stretch with Next kernel in the Armbian build tool. Thanks to shade some light here.

igorpecovnik, 2019-07-09:
> purpose of CLI_TARGET

Tells build system which targets to build. It's an internal function so its not specially documented.

> CLI_TARGET="buster,bionic:next"

... will build Debian Buser and Ubuntu Bionic with a NEXT kernel target. Whatever that target is. Currently this is 4.14.y but when this is merged, we will have NEXT with 4.19.y

Stretch with NEXT which is already present for download will automatically upgraded to 4.19.y once we release an update.

g-provost, 2019-07-09:
> Tells build system...
By build system you are talking about some nodes that are doing the nightly build ? 

> Stretch with NEXT which is already present for download will automatically upgraded to 4.19.y once we release an update.

Not sure I understand. We don't want NEXT (4.19.y) with Stretch. I was just suggesting to have DEFAULT (4.14.y) with Stretch.

igorpecovnik, 2019-07-09:
> By build system you are talking about some nodes that are doing the nightly build ?

Yes. It's when one use BUILD_ALL="yes" while for nightly CLI_BETA_TARGET variable is read. 

> We don't want NEXT (4.19.y) with Stretch. I was just suggesting to have DEFAULT (4.14.y) with Stretch.

That's upon what we will build under current build logic and later expose (just some) for download, which is set elsewhere, manually in the website management. Stretch with NEXT or DEFAULT or DEV **can** be build or user can upgrade or downgrade to it.

count-doku, 2019-07-09:
Hi there,

Testing for the clearfog pro board,
I made a fresh clone of the armbian build repo today, and using "./compile.sh EXPERT=yes LIB_TAG=mvebu" built a default, next and dev image. All using Debian Stretch. Left the kernel config untouched.

Afterwards I burned the images to a sdcard and tested them one by one. 
Seems pretty good working so far. All versions boot up, and stuff like wan, lan ports (switched) are working. From 4.19.y onward the sfp port is also working out of the box. I noticed some problems in kernel 4.19.y, my sata card did not get detected. In 5.1.y everything worked. But that might be a problem attached to loading the wrong device tree. I will conduct further testing on that tomorrow.

Attached I put an excel sheet with my exact testing configuration, used hardware and armbianmonitor links.

As a next step I will clone my current production setup and try to upgrade it to newest kernel & uboot step-by-step. Will then report back, if and when it breaks.

Let me know if you want me to test anything specific. My board also does have the spi flash.

[mvebu_testing.xlsx](https://github.com/armbian/build/files/3373717/mvebu_testing.xlsx)

Image   Name | U Boot | Kernel | Debian | Armbian | Monitor | Board (by uboot) | Boot | UART | WAN | LAN | SFP | SPI Flash | mPCIe WiFi | mPCIe SATA | m.2 / sata
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
Armbian_5.90_Clearfogpro_Debian_stretch_default_4.14.132 | 2018.01 | 4.14.132 | stretch | 5.90 | http://ix.io/1O7l | SolidRun Clearfog Pro | ok | ok | ok | ok | not working | not found | ok | ok | ok
Armbian_5.90_Clearfogpro_Debian_stretch_next_4.19.57 | 2019.04 | 4.19.57 | stretch | 5.90 | http://ix.io/1O7t | SolidRun Clearfog | ok | ok | ok | ok | ok | ok | ok | not working | ok
Armbian_5.90_Clearfogpro_Debian_stretch_dev_5.1.16 | 2019.04 | 5.1.16 | stretch | 5.90 | http://ix.io/1O7x | SolidRun Clearfog | ok | ok | ok | ok | ok | ok | ok | ok | ok
Used Hardware: |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
Clearfog Pro w/o eMMC |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
mPCIe SATA Adapter with ASM1062 Chip |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
mPCIe WLAN Adapter with QCA986X Chip   (WLE600VX) |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
SFP Cisco compatible GLC-T 1000M RJ45   Adapter |   |   |   |   |   |   |   |   |   |   |   |   |   |   |

aprayoga, 2019-07-10:
@count-doku thanks.
regarding mPCIE SATA that failed, it's a PCIE to SATA adapter, right? not mSATA.

count-doku, 2019-07-10:
Correct, using the 4.19.57 kernel the ASM1062 mPCIe to SATA adapter is not recognized in linux. But this might be because loading the wrong device tree. I think the clearfog base has only 1 pcie. I will test again later.

mSATA I cannot test, as I have no mSATA drive.

count-doku, 2019-07-10:
So back from testing. I have some bad news unfortunately. 

---

As everthing from mPCIe seemed to work ok I focused on testing the different mpcie sata devices I have (on the different kernel / uboot versions).

Tested devices:

1. mPCIe SATA Adapter with ASM1062 Chip
2. mPCIe SATA Adapter with Marvel 88SE9215 Chip

Things are not working as good as expected:
Uboot 2018 & Kernel 4.14 AND Uboot 2019 & Kernel 5.1:
After uboot, the "starting kernel" prompt hangs a long time (~2min), then the pcie devices show up in 'lspci' but the attached drives are not recognized.

```
lspci: 
00:02.0 PCI bridge: Marvell Technology Group Ltd. Device 6828 (rev 04)
00:03.0 PCI bridge: Marvell Technology Group Ltd. Device 6828 (rev 04)
01:00.0 SATA controller: ASMedia Technology Inc. ASM1062 Serial ATA Controller (rev 01)
02:00.0 SATA controller: Marvell Technology Group Ltd. Device 9215 (rev 11)

dmesg:
[    7.389247] ata5.00: qc timeout (cmd 0xec)
[    7.389255] ata5.00: failed to IDENTIFY (I/O error, err_mask=0x4)
[    7.865250] ata5: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[   17.885244] ata5.00: qc timeout (cmd 0xec)
[   17.885251] ata5.00: failed to IDENTIFY (I/O error, err_mask=0x4)
[   17.885254] ata5: limiting SATA link speed to 3.0 Gbps
[   18.361249] ata5: SATA link up 3.0 Gbps (SStatus 123 SControl 320)
[   50.141244] ata5.00: qc timeout (cmd 0xec)
[   50.141250] ata5.00: failed to IDENTIFY (I/O error, err_mask=0x4)
[   50.617249] ata5: SATA link up 3.0 Gbps (SStatus 123 SControl 320)
[   51.097250] ata6: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[   56.285245] ata6.00: qc timeout (cmd 0xec)
[   56.285252] ata6.00: failed to IDENTIFY (I/O error, err_mask=0x4)
[   56.761249] ata6: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[   66.781243] ata6.00: qc timeout (cmd 0xec)
[   66.781250] ata6.00: failed to IDENTIFY (I/O error, err_mask=0x4)
[   66.781253] ata6: limiting SATA link speed to 3.0 Gbps
[   67.265249] ata6: SATA link up 3.0 Gbps (SStatus 123 SControl 320)
[   99.293243] ata6.00: qc timeout (cmd 0xec)
[   99.293249] ata6.00: failed to IDENTIFY (I/O error, err_mask=0x4)
[   99.773248] ata6: SATA link up 3.0 Gbps (SStatus 123 SControl 320)
```

Have you changed anything in the kernel / uboot settings for pcie or sata? With the current (official) next branch everything works flawless.

Interestingly on uboot 2019 but with kernel 4.19. the mpcie sata cards don't get recognized at all.

---

I also found another uboot 2019 problem / error:
```
UBOOT 2019.1

Found U-Boot script /boot/boot.scr
2948 bytes read in 207 ms (13.7 KiB/s)
## Executing script at 03000000
Boot script loaded from mmc
256 bytes read in 167 ms (1000 Bytes/s)
19383 bytes read in 428 ms (43.9 KiB/s)
4673244 bytes read in 1041 ms (4.3 MiB/s)
5672184 bytes read in 1222 ms (4.4 MiB/s)
## Loading init Ramdisk from Legacy Image at 02880000 ...
   Image Name:   uInitrd
   Created:      2019-07-09  13:08:39 UTC
   Image Type:   ARM Linux RAMDisk Image (gzip compressed)
   Data Size:    4673180 Bytes = 4.5 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 02040000
   Booting using the fdt blob at 0x2040000
   Loading Ramdisk to 0fb8b000, end 0ffffe9c ... OK
ERROR: reserving fdt memory region failed (addr=2040000 size=6b000)
   Loading Device Tree to 0fb1d000, end 0fb8afff ... OK

UBOOT 2018 
No error, notice different address

Found U-Boot script /boot/boot.scr
2948 bytes read in 105 ms (27.3 KiB/s)
## Executing script at 00200000
Boot script loaded from mmc
172 bytes read in 84 ms (2 KiB/s)
20484 bytes read in 212 ms (93.8 KiB/s)
4683313 bytes read in 343 ms (13 MiB/s)
5455608 bytes read in 384 ms (13.5 MiB/s)
## Loading init Ramdisk from Legacy Image at 01800000 ...
   Image Name:   uInitrd
   Created:      2019-07-09   7:05:39 UTC
   Image Type:   ARM Linux RAMDisk Image (gzip compressed)
   Data Size:    4683249 Bytes = 4.5 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 00100000
   Booting using the fdt blob at 0x100000
   Loading Ramdisk to 0fb88000, end 0ffff5f1 ... OK
   reserving fdt memory region: addr=100000 size=6b000
   Loading Device Tree to 0fb1a000, end 0fb87fff ... OK
```

Notice how with uboot 2019 there is some error when reserving fdt memory region. Wrong address maybe?

---

I attached a full boot up log with uboot, kernel, dmesg. This is loading uboot 2018 with 4.14.
There are also some kernel warnings in the dmesg output. The exact position and device differs from boot to boot. They appeared for sfp, pcie, phy...

[boot.txt](https://github.com/armbian/build/files/3378314/boot.txt)

aprayoga, 2019-07-11:
> Things are not working as good as expected:
> Uboot 2018 & Kernel 4.14 AND Uboot 2019 & Kernel 5.1:
> After uboot, the "starting kernel" prompt hangs a long time (~2min), then the pcie devices show up in 'lspci' but the attached drives are not recognized.

I wanted to confirm, both configuration/combination has this same issue, right?

> Have you changed anything in the kernel / uboot settings for pcie or sata? With the current (official) next branch everything works flawless.

The current official next is LK 4.14 with U-Boot version 2013.01, is it?

> Interestingly on uboot 2019 but with kernel 4.19. the mpcie sata cards don't get recognized at all.
> 
I'll check whether the driver compiled or maybe compiled but somehow not loaded automatically. 

---
> I also found another uboot 2019 problem / error:
> 
> ```
> UBOOT 2019.1
>
> ## Flattened Device Tree blob at 02040000
>    Booting using the fdt blob at 0x2040000
>    Loading Ramdisk to 0fb8b000, end 0ffffe9c ... OK
> ERROR: reserving fdt memory region failed (addr=2040000 size=6b000)
>    Loading Device Tree to 0fb1d000, end 0fb8afff ... OK
> ```
> Notice how with uboot 2019 there is some error when reserving fdt memory region. Wrong address maybe?

Yes this error appear after commit cd2429c. This changes needed to be able to run bootscript based on [boot-marvell.cmd](https://github.com/armbian/build/blob/master/config/bootscripts/boot-marvell.cmd) due to these lines:
```
setenv fdt_high 0xffffffff
setenv initrd_high 0xffffffff
```
If the load address is not updated, it would failed to boot.
This is more concerned for existing user. For new user /new sd card image, we don't need this changes as it already based on [boot-mvebu-next.cmd](https://github.com/armbian/build/blob/master/config/bootscripts/boot-mvebu-next.cmd). On Helios4, we had trouble during Armbian 5.68-5.70 where we upgraded the U-boot from 2013 to mainline 2018 because of this incompatibility, the system no longer boot.
I tested on Helios4 and nothing affected so i dismissed this more as warning, maybe I should have compare what in RAM and on the file.

count-doku, 2019-07-11:
> I wanted to confirm, both configuration/combination has this same issue, right?

Yes the mpcie issue is the same. 

> The current official next is LK 4.14 with U-Boot version 2013.01, is it?

Yes, more specifically LK4.14.121.

Regarding uboot fdt memory region error, maybe this is a mere warning and has no influence on the other things noticed.

---

As a next step to further narrow down the mpcie problem would a be a viable step if I took my running LK4.14.121 / uboot 2013.1 install and updated only the uboot to 2018? Or just leave the uboot and update the kernel to yours (lk4.14.123 I think)?
That way we could confirm wether this issue is uboot or kernel related.

aprayoga, 2019-07-11:
> As a next step to further narrow down the mpcie problem would a be a viable step if I took my running LK4.14.121 / uboot 2013.1 install and updated only the uboot to 2018? Or just leave the uboot [...]

I'm more suspicious to u-boot since it's the one who configured the SerDes, so i think leave the kernel and update the u-boot to 2018

count-doku, 2019-07-11:
Didn't have much time left this morning, so I just cloned my production setup which is uboot 2013.1 with lk4.14.121 and upgraded uboot to 2018.01.

Doesn't boot at all. See log below:
```
U-Boot SPL 2018.01-armbian (Jul 09 2019 - 08:32:52)
High speed PHY - Version: 2.0
Detected Device ID 6828
board SerDes lanes topology details:
 | Lane #  | Speed |  Type       |
 --------------------------------
 |   0    |  3   |  SATA0       |
 |   1    |  0   |  SGMII1      |
 |   2    |  5   |  PCIe1       |
 |   3    |  5   |  USB3 HOST1  |
 |   4    |  5   |  PCIe2       |
 |   5    |  0   |  SGMII2      |
 --------------------------------
:** Link is Gen1, check the EP capability
PCIe, Idx 1: Link upgraded to Gen2 based on client capabilities
PCIe, Idx 2: detected no link
High speed PHY - Ended Successfully
DDR3 Training Sequence - Ver TIP-1.29.0
DDR3 Training Sequence - Switching XBAR Window to FastPath Window
DDR3 Training Sequence - Ended Successfully
Trying to boot from MMC1

U-Boot 2018.01-armbian (Jul 09 2019 - 08:32:52 +0200)

SoC:   MV88F6828-A0 at 1600 MHz
DRAM:  1 GiB (800 MHz, 32-bit, ECC not enabled)
MMC:   mv_sdh: 0
*** Warning - bad CRC, using default environment

Model: SolidRun Clearfog
Board: SolidRun ClearFog Pro
SCSI:  AHCI 0001.0000 32 slots 2 ports 6 Gbps 0x3 impl SATA mode
Net:
Warning: ethernet@70000 (eth1) using random MAC address - d2:63:b4:2d:9d:71
eth1: ethernet@70000
Warning: ethernet@30000 (eth2) using incremented MAC address - d2:63:b4:2d:9d:72
, eth2: ethernet@30000
Warning: ethernet@34000 (eth3) using incremented MAC address - d2:63:b4:2d:9d:73
, eth3: ethernet@34000
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot/boot.scr
1604 bytes read in 97 ms (15.6 KiB/s)
## Executing script at 00200000
Boot script loaded from mmc
load - load binary file from a filesystem

Usage:
load <interface> [<dev[:part]> [<addr> [<filename> [bytes [pos]]]]]
    - Load binary file 'filename' from partition 'part' on device
       type 'interface' instance 'dev' to address 'addr' in memory.
      'bytes' gives the size to load in bytes.
      If 'bytes' is 0 or omitted, the file is read until the end.
      'pos' gives the file byte position to start reading from.
      If 'pos' is 0 or omitted, the file is read from the start.
load - load binary file from a filesystem

Usage:
load <interface> [<dev[:part]> [<addr> [<filename> [bytes [pos]]]]]
    - Load binary file 'filename' from partition 'part' on device
       type 'interface' instance 'dev' to address 'addr' in memory.
      'bytes' gives the size to load in bytes.
      If 'bytes' is 0 or omitted, the file is read until the end.
      'pos' gives the file byte position to start reading from.
      If 'pos' is 0 or omitted, the file is read from the start.
4709951 bytes read in 341 ms (13.2 MiB/s)
5464240 bytes read in 382 ms (13.6 MiB/s)
## Loading init Ramdisk from Legacy Image at 01800000 ...
   Image Name:   uInitrd
   Created:      2019-07-07  16:49:18 UTC
   Image Type:   ARM Linux RAMDisk Image (gzip compressed)
   Data Size:    4709887 Bytes = 4.5 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.

Error: unrecognized/unsupported machine ID (r1 = 0x00000000).

Available machine support:

ID (hex)        NAME
ffffffff        Generic DT based system
ffffffff        Marvell Armada 39x (Device Tree)
ffffffff        Marvell Armada 380/385 (Device Tree)
ffffffff        Marvell Armada 375 (Device Tree)
ffffffff        Marvell Armada 370/XP (Device Tree)
ffffffff        Marvell Dove

Please check your kernel config and/or bootloader.
```

Might try again this afternoon with a fresh image from dl.armbian.com and upgrade from there.

aprayoga, 2019-07-11:
This is the incompatibility issue I mentioned earlier.
could you try
```
setenv boot_a_script 'load ${devtype} ${devnum}:${distro_bootpart} ${scriptaddr} ${prefix}${script}; setenv boot_interface ${devtype};source ${scriptaddr}'
setenv loadaddr 0x02000000
setenv fdt_addr ${fdt_addr_r}
saveenv
```
on U-Boot prompt. It supposed to fix it and the system can boot.

count-doku, 2019-07-11:
What should I run after 'saveenv'?

I tried 'run boot_a_script':
```=> run boot_a_script
** Bad device : 0x200000 **
## Executing script at 00200000
Wrong image format for "source" command
```

And just the normal 'boot' and 'reset':
```=> boot
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot/boot.scr
1604 bytes read in 97 ms (15.6 KiB/s)
## Executing script at 00200000
Boot script loaded from mmc
221 bytes read in 85 ms (2 KiB/s)
20596 bytes read in 223 ms (89.8 KiB/s)
4709951 bytes read in 339 ms (13.2 MiB/s)
5464240 bytes read in 382 ms (13.6 MiB/s)
## Loading init Ramdisk from Legacy Image at 01800000 ...
   Image Name:   uInitrd
   Created:      2019-07-07  16:49:18 UTC
   Image Type:   ARM Linux RAMDisk Image (gzip compressed)
   Data Size:    4709887 Bytes = 4.5 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 00100000
   Booting using the fdt blob at 0x100000
   Using Device Tree in place at 00100000, end 00108073

Starting kernel ...

Uncompressing Linux... done, booting the kernel.

Error: unrecognized/unsupported machine ID (r1 = 0x00000000).

Available machine support:

ID (hex)        NAME
ffffffff        Generic DT based system
ffffffff        Marvell Armada 39x (Device Tree)
ffffffff        Marvell Armada 380/385 (Device Tree)
ffffffff        Marvell Armada 375 (Device Tree)
ffffffff        Marvell Armada 370/XP (Device Tree)
ffffffff        Marvell Dove

Please check your kernel config and/or bootloader.
```
---

Will test some more and edit this once done.

count-doku, 2019-07-17:
Hey,

as I received the boards today, I powered up the clearfog pro and tried each kernel / uboot variant again. Biggest noted difference: Correct board detection, thanks @aprayoga. Saw your commit, works.

Apart from that, I used Debian Buster. Works like a charm, no (new) problems compared to Stretch!

SFP and SPI Flash are both working going forward from Kernel 4.19 as expected. M.2 SSD is working always. WAN and switched LAN ports are also fine.

Biggest issue is still the mPCIe recognition. I was unable to get the extender cards to really work. Same problems as before (obviously, as nothing was changed. With the wifi cards the driver doesn't get loaded correctly (iw list shows no devices) - SATA card gets detected on 4.14. and 5.xx but is not working with attached drives.

How are we going to procede here? I can leave a board powered up, and give you ssh access if you need to check out some logs or so. Or do you have an idea already what might be wrong?

I think we should concentrate on getting the boards to work with new images first. Afterwards think about the update route from old versions.

Attached logs of current boot, dmesg, etc. If you need anything specific just let me know.
[Armbian_5.90_Clearfogpro_Debian_buster_default_4.14.133.txt](https://github.com/armbian/build/files/3403758/Armbian_5.90_Clearfogpro_Debian_buster_default_4.14.133.txt)
[Armbian_5.90_Clearfogpro_Debian_buster_dev_5.1.18.txt](https://github.com/armbian/build/files/3403759/Armbian_5.90_Clearfogpro_Debian_buster_dev_5.1.18.txt)
[Armbian_5.90_Clearfogpro_Debian_buster_next_4.19.59.txt](https://github.com/armbian/build/files/3403760/Armbian_5.90_Clearfogpro_Debian_buster_next_4.19.59.txt)

---

I will try the clearfog base tomorrow, and maybe on the weekend put together some testbed with an relais to powercycle the boards and some additional sbc to access the serial ports via ssh.

g-provost, 2019-07-18:
Let's focus on Default and Next, and put aside Dev for now. 

Can you provide a dedicated list of what works and doesn't work for each build (Default and Next).

Can you provide a picture of your setup ? How do you power supply the SSD/HDD connected to the mPCie SATA controller (Marvell 88SE9215) ?

aprayoga, 2019-07-18:
> How are we going to procede here? I can leave a board powered up, and give you ssh access if you need to check out some logs or so. Or do you have an idea already what might be wrong?

Unfortunately I still have no idea what might be wrong. This morning I tried to see whether there is any configuration (in PCI subsystem) differences between u-boot 2013 and 2018. But on u-boot 2018 the pci command is not enabled and when i enabled it together with PCI support, got some compile error.

> put together some testbed with an relais to powercycle the boards and some additional sbc to access the serial ports via ssh.

That would be nice

And regarding the phy error on LK 4.14, do you encounter the same issue with Armbian_5.75_Clearfogpro_Debian_stretch_next_4.14.98 ?

igorpecovnik, 2019-07-18:
> when i enabled it together with PCI support, got some compile error

Perhaps @rabeeh have some ideas what we are missing here?

count-doku, 2019-07-18:
> Let's focus on Default and Next, and put aside Dev for now.

Ok. I'd go even further and say lets get default sorted out first. Because there might be a mess bigger than expected.

> Can you provide a dedicated list of what works and doesn't work for each build (Default and Next).

Is listing this here ok, or did you mean some google docs sheet or so?

Default: 
- [x] Boot via sd card, with buster
- [x] lan and wan port
- [x] m.2
- [x] usb
- [x] cpu freq stepping (between 800MHz and 1.6 GHz)
- [x] all the other things that worked before ;) 
- [ ] spi flash (does not appear in linux at all)
- [ ] sfp (but I guess this cannot work here without backporting alot)
- [ ] mpcie sata card (this gets detected fine, but attached drives don't)
- [ ] wifi card (this gets detected fine, but driver is not loaded. )

Next: 
- [x] Boot via sd card, with buster
- [x] lan and wan port
- [x] m.2
- [x] usb
- [x] cpu freq stepping (between 800MHz and 1.6 GHz)
- [x] sfp working very good. Including dhcp etc.
- [x] spi flash (read / write tested)
- [ ] mpcie sata card
- [ ] wifi card

I focused on default now.

- Regarding wifi: 
Requires installation of 'firmware-atheros' (did so on older versions as well)
After that it works on LK4.14.98 but on LK4.14.113 dmesg gives:

```
[    7.824666] pci 0000:00:03.0: enabling device (0140 -> 0142)
[    7.824685] ath10k_pci 0000:02:00.0: enabling device (0140 -> 0142)
[    7.824929] ath10k_pci 0000:02:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[    7.849390] zram1: detected capacity change from 0 to 526585856
[    8.099943] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/pre-cal-pci-0000:02:00.0.bin failed with error -2
[    8.099948] ath10k_pci 0000:02:00.0: Falling back to user helper
[    8.389518] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/cal-pci-0000:02:00.0.bin failed with error -2
[    8.389523] ath10k_pci 0000:02:00.0: Falling back to user helper
[    8.394551] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/QCA988X/hw2.0/firmware-6.bin failed with error -2
[    8.394555] ath10k_pci 0000:02:00.0: Falling back to user helper
[    8.407939] ath10k_pci 0000:02:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043222ff sub 0000:0000
[    8.407947] ath10k_pci 0000:02:00.0: kconfig debug 0 debugfs 0 tracing 0 dfs 0 testmode 0
[    8.408484] ath10k_pci 0000:02:00.0: firmware ver 10.2.4-1.0-00041 api 5 features no-p2p,raw-mode,mfp,allows-mesh-bcast crc32 f43fa422
[    8.454013] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/QCA988X/hw2.0/board-2.bin failed with error -2
[    8.454021] ath10k_pci 0000:02:00.0: Falling back to user helper
[    8.459085] ath10k_pci 0000:02:00.0: board_file api 1 bmi_id N/A crc32 bebc7c08
[    8.909405] Adding 514240k swap on /dev/zram1.  Priority:5 extents:1 across:514240k SSFS
[    9.138638] zram0: detected capacity change from 0 to 52428800
[   10.230400] random: crng init done
[   10.230407] random: 7 urandom warning(s) missed due to ratelimiting
[   10.432337] EXT4-fs (zram0): mounted filesystem without journal. Opts: discard
[   11.132892] ath10k_pci 0000:02:00.0: failed to receive control response completion, polling..
[   12.156891] ath10k_pci 0000:02:00.0: Service connect timeout
[   12.156898] ath10k_pci 0000:02:00.0: failed to connect htt (-110)
[   12.239842] ath10k_pci 0000:02:00.0: could not init core (-110)
[   12.239877] ath10k_pci 0000:02:00.0: could not probe fw (-110)
```

Device does not appear in 'iw list' then.

- Regarding 88se9215:
Works on LK 4.14.98; can not read drives on LK 4.14.113 / uboot2018
[... 27 lines cut ...]

--- 
**BUT** and this is a big but. 

The errors above are only there, if only on card is plugged in. So either sata or wifi. BUT if I plug in both cards bigger problems arise. 

If I plug in wifi into the pcie closest to the cpu, only the wifi one gets detected. Not the sata adapter.
If I reverse the order, so sata is closest to cpu and wifi farthest. Nothing gets detected at all.
Above points are about linux detection, uboot always shows both devices.

But this is another problem that exists longer already, this is also true for LK 4.14.98. There even was some forum post about this a while back (here: https://forum.armbian.com/topic/9291-clearfogpro-lost-pcie-sata-card-after-update/) where we unfortunately told the guy that it has to be a problem on his end. It seems like it isn't... I will answer the forum post about that.

---

> Can you provide a picture of your setup ? How do you power supply the SSD/HDD connected to the mPCie SATA controller (Marvell 88SE9215) ?

Using the same power supply as for the clearfog (12V 4A) and for the 5V rail I use some adjustable buck converter. Voltages are ok at all components (measured with oscilloscope, voltage and ripple is fine). The hdd is also running. 
This is also working fine (hdd gets detected) with LK4.14.98 (Armbian 5.75)

[image]

> And regarding the phy error on LK 4.14, do you encounter the same issue with 
> Armbian_5.75_Clearfogpro_Debian_stretch_next_4.14.98 ?

No phy error here.

count-doku, 2019-07-19:
Tested some more. To minimize confusion and possible side effects I built a fresh image from armbian/mvebu default branch (uboot 2018.1, lk 4.14.133).

I booted the clearfogpro without any pci devices, installed the 'firmware-atheros' package via apt, then I tried all possible configurations and recorded uboot, lspci and dmesg output. 

To clarify, PCIe Idx2 is the slot closest to the cpu. 
[image]

The issue got a bit better (why ever?). Now atleast the cards are not 'hiding' one another in linux. 

Checkout the table and attached logs.

  |   | Uboot | lspci | dmesg
-- | -- | -- | -- | --
no card |   | detected no link, both slots | just the pci bridges |  
  |   |   |   |  
Wifi in Idx1 |   | detected no link, both slots | found in  idx1 | driver loads, but cannot connect
Wifi in Idx2 |   | detects idx2 | found in  idx2 | driver loads, but cannot connect
  |   |   |   |  
Sata in Idx1 |   | detected nothing in idx2 (correct); no message regarding idx1 | found in idx1 | driver loads, attached device fails
Sata in Idx2 |   | detected nothing in idx1 (correct); no message regarding idx2 | found in idx2 | driver loads, attached device fails
  |   |   |   |  
Both with: |   |   |   |  
sata in 1; wifi in 2 |   | detected nothing in idx2; no message regarding idx1 | found both correctly | same as above, does not work
sata in 2; wifi in 1 |   | detected nothing in idx1; no message regarding idx2 | found both correctly | same as above, does not work

[no_pcie.txt](https://github.com/armbian/build/files/3409942/no_pcie.txt)

[wifi_in_idx1.txt](https://github.com/armbian/build/files/3409943/wifi_in_idx1.txt)
[wifi_in_idx2.txt](https://github.com/armbian/build/files/3409945/wifi_in_idx2.txt)

[sata_in_idx1.txt](https://github.com/armbian/build/files/3409944/sata_in_idx1.txt)
[sata_in_idx2.txt](https://github.com/armbian/build/files/3409946/sata_in_idx2.txt)

[wifi_in_idx2_and_sata_in_idx1.txt](https://github.com/armbian/build/files/3409947/wifi_in_idx2_and_sata_in_idx1.txt)
[sata_in_idx2_and_wifi_in_idx1.txt](https://github.com/armbian/build/files/3409941/sata_in_idx2_and_wifi_in_idx1.txt)

g-provost, 2019-07-19:
Can you provide same test with Armbian_5.75_Clearfogpro_Debian_stretch_next_4.14.98 in order to compare.

rabeeh, 2019-07-19:
Hi,

Can you please double check that the pcie Force receiver detect patch is
applied?

https://github.com/SolidRun/lede-project/blob/sr-testing/package/boot/uboot-mvebu/patches/0014-mvebu-a38x-Force-receiver-detected-on-PCIe-lanes.patch

A similar patch was applied by Baruch to u-boot mainline few months ago.

count-doku, 2019-07-20:
@aprayoga and I did further debugging today. He found the bug in uboot: 
"The issue for SATA, actually simple. the u-boot 2018.01 is not compiled with PCI support.
I enabled it and recompiled"
Marvell 88SE9215 (sata expander) is now working, but as I understand it, there is more work to be done here. 
"now my concern is,
on u-boot 2018 onward, the board configuration are using device tree.
under PCIe node there is properties of reset-gpio. On current driver i don't see the reset gpio being use.
in u-boot 2013, the gpio is executed under custom board code
I will try to modify the u-boot 2018 driver to make use of this gpio" (aprayoga)

The wifi card (wle600vx, qca988x) still has problems loading the driver, still trying to figure that out.
Intel wifi cards are already working (intel ac-3160 & intel n-2200 tested), so this does not seem to be a general problem.

count-doku, 2019-07-22:
Good news. After aprayoga pushed his uboot changes, I build new images for Clearfog Base & Pro, default and next options. 

With default (Uboot 2018.1 and Kernel 4.14.133) about everything that can be expected is working.

---

Default: 
- [x] Boot via sd card, with buster
- [x] lan and wan port
- [x] m.2
- [x] usb
- [x] cpu freq stepping (between 800MHz and 1.6 GHz)
- [x] all the other things that worked before ;) 
- [x] mpcie sata card (tested marvell 88se9215 and asmedia 1061)
- [x] mpcie wifi card (tested wle600vx (qca988x) and intel ones)
- [ ] spi flash (does not appear in linux at all)
- [ ] sfp (but I guess this cannot work here without backporting alot)

I guess this is all that can be made working with reasonable amount of time. Everything above was tested in clearfogpro and base and the pcie cards in every possible combination.

-> This can be shipped.

Next: 
- [x] Boot via sd card, with buster
- [x] lan and wan port
- [x] m.2
- [x] usb
- [x] cpu freq stepping (between 800MHz and 1.6 GHz)
- [x] sfp working very good. Including dhcp etc.
- [x] spi flash (read / write tested)
- [ ] mpcie sata card
- [ ] wifi card

With next unfortunately the mcpie is not working (same problem left as before with default). I attached full log. 

[cfplog.txt](https://github.com/armbian/build/files/3418954/cfplog.txt)

---

I did not test mSATA because I am unsure how to switch pcie to msata. The documentation has a dead link only. https://docs.armbian.com/Hardware_Marvell/

igorpecovnik, 2019-07-23:
> This can be shipped.

OK, I will make images once merged to master.

> With next unfortunately the mcpie is not working

Since mpci is most important feature on this board ... Can you check how good is 5.x.y out of the box?

g-provost, 2019-07-23:
@igorpecovnik Please let @aprayoga gives green light when everything is ready to be built for mvebu.

g-provost, 2019-07-23:
@count-doku You said via PM that you have another mPCIe  SATA controller (based on ASM1061), do you also see the same issue under Next build with the ASM1061 controller ?

Here I attach an email reply from Baruch which is linked to the issue. Might no be useful but for the sake of tracking better to attach it :

> There is a known issue with the Marvell PCIe SATA controller on 32-bit
> Armada 38x. The same controller works fine with Armada 8040 on Clearfog
> GT-8K. I suspect that the driver has a 32-bit compatibility issue
> somewhere.
> 
> An ASMedia PCIe SATA controller I tested works fine on Clearfog.
> 
> The Helios4 board uses no PCIe interface. All SATA disks are attached to
> the native Armada 388 SATA controller.
> 
> I tested the same WLE900VX card with ASMedia SATA controller attached to
> both mini-PCIe slots on Clearfog Pro. Both cards were detected. The SATA
> controller seems to probe successfully. The WLE900VX card driver shows
> the same error messages shown in the Armbian bug report.
> 
> [    2.124633] ath10k_pci 0000:02:00.0: enabling device (0140 -> 0142)
> [    2.131140] ath10k_pci 0000:02:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
> [    2.362597] ath10k_pci 0000:02:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000
> [    2.371881] ath10k_pci 0000:02:00.0: kconfig debug 0 debugfs 0 tracing 0 dfs 0 testmode 0
> [    2.380861] ath10k_pci 0000:02:00.0: firmware ver 10.2.4-1.0-00041 api 5 features no-p2p,raw-mode,mfp,allows-mesh-bcast crc32 f43fa422
> [    2.425709] ath10k_pci 0000:02:00.0: board_file api 1 bmi_id N/A crc32 bebc7c08
> [    4.566986] ath10k_pci 0000:02:00.0: failed to receive control response completion, polling..
> [    5.616981] ath10k_pci 0000:02:00.0: Service connect timeout
> [    5.622661] ath10k_pci 0000:02:00.0: failed to connect htt (-110)
> [    5.757716] ath10k_pci 0000:02:00.0: could not init core (-110)
> [    5.763677] ath10k_pci 0000:02:00.0: could not probe fw (-110)

[31 more replies]

closed 2019-08-01 by igorpecovnik