armbian/build #1429: A388 SOM U-Boot ODT Fix

aprayoga, 2019-06-27:
According to SolidRun, old versions of U-Boot did not configure correctly the ODT on data signals of DDR RAM on ARMADA A388 SOMs. All systems with ARMADA A388 SOM are affected. Actual corruption has only been observed on SOM rev 2.1.

URL:
https://developer.solid-run.com/knowledge-base/armada-38x-som-u-boot-odt-update/

aprayoga, 2019-06-27:
SolidRun U-Boot already fixed on their repo.
Helios4 U-Boot based on upstream version 2018.11 is fixed since commit https://github.com/helios-4/u-boot/commit/8f200a3d28e22ff9d5340f1a0493e0561d168069

wpiecek, 2019-06-27:
> helios-4/u-boot@8f200a3

Hello,

id there any working image for Banana Pi M2 Zero?
I identify generally two versions - one, with working wifi but not working
GPIO (I build BPI, but still not recognized the board). Second have working
GPIO but network is not stable (crash after ~~1h). First is based on kernel
4.19 and second on 3.4.

Is there any other option than build them from the scratch?

aprayoga, 2019-07-01:
Testing on SOM rev 2.1

aprayoga, 2019-07-01:
According to SolidRun's [A388 SOM Product Change Notification](https://developer.solid-run.com/wp-content/uploads/2019/05/SR-A388-SOM-PCN-200519.pdf), in addition of ODT update, SOM rev 2.1 also change the silicon revision from A0 to B0.

Following log captured on Helios4 with ODT update
```
U-Boot 2018.11-armbian (Jul 01 2019 - 16:28:58 +0800)

SoC:   MV88F6828-?? (a) at 1600 MHz
DRAM:  2 GiB (800 MHz, 32-bit, ECC enabled)
MMC:   mv_sdh: 0
Loading Environment from MMC... *** Warning - bad CRC, using default environment
```

_Add additional task to do cosmetic fix to recognise B0 silicon version._

aprayoga, 2019-07-02:
Tested using U-Boot command:
`mtest 800000 7f77ffff`

on A388 SOM Rev 2.1 (2GB RAM ECC) using Helios4 board.
No memory corruption observed during the test, with or without ODT update.

aprayoga, 2019-07-02:
Helios4 U-Boot v2018.11 recognise B0 revision
```
U-Boot SPL 2018.11-armbian (Jul 02 2019 - 01:29:11 +0800)
High speed PHY - Version: 2.0
Detected Device ID 6828
board SerDes lanes topology details:
 | Lane #  | Speed |  Type       |
 --------------------------------
 |   0    |  6   |  SATA0       |
 |   1    |  5   |  USB3 HOST0  |
 |   2    |  6   |  SATA1       |
 |   3    |  6   |  SATA3       |
 |   4    |  6   |  SATA2       |
 |   5    |  5   |  USB3 HOST1  |
 --------------------------------
High speed PHY - Ended Successfully
mv_ddr: mv_ddr-armada-17.10.4
DDR3 Training Sequence - Switching XBAR Window to FastPath Window
DDR Training Sequence - Start scrubbing
DDR3 Training Sequence - End scrubbing
mv_ddr: completed successfully
Trying to boot from MMC1

U-Boot 2018.11-armbian (Jul 02 2019 - 01:29:11 +0800)

SoC:   MV88F6828-B0 at 1600 MHz
DRAM:  2 GiB (800 MHz, 32-bit, ECC enabled)
MMC:   mv_sdh: 0
Loading Environment from MMC... *** Warning - bad CRC, using default environment

Model: Helios4
Board: Helios4
SCSI:  MVEBU SATA INIT
SATA link 0 timeout.
SATA link 1 timeout.
AHCI 0001.0000 32 slots 2 ports 6 Gbps 0x3 impl SATA mode
flags: 64bit ncq led only pmp fbss pio slum part sxs

Net:
Warning: ethernet@70000 (eth1) using random MAC address - 7a:7a:fc:e7:71:7a
eth1: ethernet@70000
Hit any key to stop autoboot:  0
=>
```

SolidRun U-Boot v2018.01 recognise B0 revision
```
U-Boot SPL 2018.01-armbian (Jul 02 2019 - 10:17:53)
High speed PHY - Version: 2.0
Detected Device ID 6828
board SerDes lanes topology details:
 | Lane #  | Speed |  Type       |
 --------------------------------
 |   0    |  3   |  SATA0       |
 |   1    |  0   |  SGMII1      |
 |   2    |  5   |  PCIe1       |
 |   3    |  5   |  USB3 HOST1  |
 |   4    |  5   |  PCIe2       |
 |   5    |  0   |  SGMII2      |
 --------------------------------
PCIe, Idx 1: detected no link
PCIe, Idx 2: detected no link
High speed PHY - Ended Successfully
DDR3 Training Sequence - Ver TIP-1.29.0
DDR3 Training Sequence - Switching XBAR Window to FastPath Window
DDR3 Training Sequence - Ended Successfully
Trying to boot from MMC1

U-Boot 2018.01-armbian (Jul 02 2019 - 10:17:53 +0800)

SoC:   MV88F6828-B0 at 1600 MHz
DRAM:  1 GiB (800 MHz, 32-bit, ECC not enabled)
MMC:   mv_sdh: 0
*** Warning - bad CRC, using default environment

Model: SolidRun Clearfog
Board: SolidRun ClearFog Pro
SCSI:  SATA link 0 timeout.
AHCI 0001.0000 32 slots 2 ports 6 Gbps 0x3 impl SATA mode
Net:
Warning: ethernet@70000 (eth1) using random MAC address - d2:63:b4:b7:5e:3c
eth1: ethernet@70000
Warning: ethernet@30000 (eth2) using incremented MAC address - d2:63:b4:b7:5e:3d
, eth2: ethernet@30000
Warning: ethernet@34000 (eth3) using incremented MAC address - d2:63:b4:b7:5e:3e
, eth3: ethernet@34000
Hit any key to stop autoboot:  0
=>
```

closed 2019-07-02 by aprayoga