armbian/build #1437: Aria2 is not auto recognize proxy.

1455294568, 2019-06-30:
Aria2 is not auto recognize proxy., then the speed is going down to 20kb/s,  please use wget instead.

igorpecovnik, 2019-06-30:
Huh. @mzhboy Shell we put aria2 under a switch and make wget default? Is upstream version in a better shape than the one shipped with Bionic?

lanefu, 2019-07-09:
@1455294568 

How are you defining the proxy in your environment?

https://aria2.github.io/manual/en/html/aria2c.html#environment

zhangn1985, 2019-07-09:
if you use docker build, please add http_proxy to ENV line in Dockerfile

igorpecovnik, 2019-07-09:
Any more elegant way to solve this? Or putting at least some note to the right place?

zhangn1985, 2019-07-10:
@igorpecovnik 

>   -e, --env list                       Set environment variables
>       --env-file list                  Read in a file of environment variables

could we use this option to run container? thus could automatically grab proxy settings from shell env.

1455294568, 2019-08-11:
> @1455294568
> 
> How are you defining the proxy in your environment?
> 
> https://aria2.github.io/manual/en/html/aria2c.html#environment

export http_proxy=xxx
export https_proxy=xxx

it works on wget, but it seems like it doesn't work on aria2

igorpecovnik, 2023-03-01:
Not used anymore, closing.

closed 2023-03-01 by igorpecovnik